ondrejvysek/HomeAssistant-CEZDistribuce-PND #1: 'utf-8' codec can't decode byte 0xec in position 69: invalid continuation byte

AdamGrumlik, 2024-05-06:
Ahoj, v /homeassistant/appdaemon/apps/pnd/*.csv mám chybu:
'utf-8' codec can't decode byte 0xec in position 69: invalid continuation byte
čím to může být?

Děkuji.

ondrejvysek, 2024-05-06:
Soubory jsou kodované v latin1 a File editor je neumí korekně zobrazit - zkusit stáhnout a zobrazit mimo File file editor (Notepad, Notepad++ či jiné). Vidím tam to samé, nicméně vše funguje jak má.

AdamGrumlik, 2024-05-06:
> Soubory jsou kodované v latin1 a File editor je neumí korekně zobrazit - zkusit stáhnout a zobrazit mimo File file editor (Notepad, Notepad++ či jiné). Vidím tam to samé, nicméně vše funguje jak má.

OK, děkuji. Stáhl jsem a data tam jsou, takže skvělé.

closed 2024-05-06 by ondrejvysek