ondrejvysek/HomeAssistant-CEZDistribuce-PND #3: Unexpected error

Richmund7, 2024-05-07:
Kde delam chybu prosim?

2024-05-07 09:16:06.281879 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-05-07 09:16:06.282238 INFO AppDaemon: Python version is 3.11.9
2024-05-07 09:16:06.282567 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-05-07 09:16:06.282858 INFO AppDaemon: Added log: AppDaemon
2024-05-07 09:16:06.283090 INFO AppDaemon: Added log: Error
2024-05-07 09:16:06.283282 INFO AppDaemon: Added log: Access
2024-05-07 09:16:06.283454 INFO AppDaemon: Added log: Diag
2024-05-07 09:16:06.283684 INFO AppDaemon: Added log: pnd
2024-05-07 09:16:06.596176 INFO AppDaemon: Initializing HTTP
2024-05-07 09:16:06.597010 INFO AppDaemon: Using 'ws' for event stream
2024-05-07 09:16:06.602059 INFO AppDaemon: Starting API
2024-05-07 09:16:06.605974 INFO AppDaemon: Starting Admin Interface
2024-05-07 09:16:06.607017 INFO AppDaemon: Starting Dashboards
2024-05-07 09:16:06.615047 WARNING AppDaemon: ------------------------------------------------------------
2024-05-07 09:16:06.615693 WARNING AppDaemon: Unexpected error during run()
2024-05-07 09:16:06.616267 WARNING AppDaemon: ------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/__main__.py", line 160, in run
    loop.run_until_complete(asyncio.gather(*pending))
  File "/usr/lib/python3.11/asyncio/base_events.py", line 654, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/admin_loop.py", line 18, in loop
    await self.AD.threading.get_callback_update()
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 82, in get_callback_update
    await self.set_state("_threading", "admin", "sensor.callbacks_average_fired", state=fired_avg)
  File "/usr/lib/python3.11/site-packages/appdaemon/state.py", line 632, in set_state
    plugin = await self.AD.plugins.get_plugin_object(namespace)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/plugin_management.py", line 220, in get_plugin_object
    for name in self.plugins:
TypeError: 'NoneType' object is not iterable
2024-05-07 09:16:06.623919 INFO AppDaemon: Previous message repeated 1 times
2024-05-07 09:16:06.624488 INFO AppDaemon: AppDaemon Exited

ondrejvysek, 2024-05-07:
To vypadá na chybu při startu AppDaemon - prosím:
* kontrolu modulů v AppDaemon
* kontrolu appdaemon.yaml, především pak url k homeassistant a token (pozor v yaml musí být mezery, nikoliv tab)
* o snip obrazovky s konfigurací AppDaemon a appdaemon.yaml

Richmund7, 2024-05-07:
Diky za reakci. Zkusil jsem reinstall AppDaemon. Nevim kde zkontrolovat ty moduly?
Ted to vypada nasledovne. Prikladam oba logy a 2x snip:
2024-05-07 14:12:02.222068 INFO pnd: >>>>>>>>>>>> PND Run Event
2024-05-07 14:12:02.223860 WARNING pnd: pnd: Entity binary_sensor.pnd_running not found in namespace default
2024-05-07 14:12:02.239532 INFO pnd: ----------------------------------------------
2024-05-07 14:12:02.242090 INFO pnd: Hello from AppDaemon for Portal Namerenych Dat
2024-05-07 14:12:03.346600 INFO pnd: Driver Loaded
2024-05-07 14:12:05.606016 INFO pnd: Website Opened
2024-05-07 14:12:19.484996 INFO pnd: H1 tag with text 'Nam����en�� data' is present.
2024-05-07 14:12:32.688218 INFO pnd: Rychla Sestava selected successfully!

2024-05-07 14:12:02.224979 INFO AppDaemon: pnd: Entity binary_sensor.pnd_running created in namespace: default
2024-05-07 14:12:36.090817 WARNING pnd: ------------------------------------------------------------
2024-05-07 14:12:36.091757 WARNING pnd: Unexpected error in worker for App pnd:
2024-05-07 14:12:36.092502 WARNING pnd: Worker Ags: {'id': 'c61aece231184d4db76e3f0323df7517', 'name': 'pnd', 'objectid': 'c6c12aa436c94198833f0a9681120bcf', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f9cad96d0>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2024-05-07T12:12:02.201446+00:00', 'context': {'id': '01HX9FFJRPFZKF64PZNHA6PC56', 'parent_id': '01HX9FFJRNV6KMKQV6QH75Q44K', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2024-05-07 14:12:36.093211 WARNING pnd: ------------------------------------------------------------
2024-05-07 14:12:36.099488 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/homeassistant/appdaemon/apps/pnd.py", line 182, in run_pnd
    option = wait.until(EC.visibility_of_element_located((By.XPATH, f"//li[contains(., '{self.EAN}')]")))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/support/wait.py", line 105, in until
    raise TimeoutException(message, screen, stacktrace)
selenium.common.exceptions.TimeoutException: Message: 

2024-05-07 14:12:36.100344 WARNING pnd: ------------------------------------------------------------

[image]

[image]

ondrejvysek, 2024-05-07:
To vypadá mnohem lépe, script se spustil a umřel na vyhledání EAN na portále - je v konfiguraci aplikace EAN? Je tento EAN vidět na portále?

Richmund7, 2024-05-07:
Dekuji za navigaci. Nechtelo to EAN, ale cislo elektromeru:
[image]

nabehli entity, jen interval consumption = interval production. Kde muze byt problem prosim?
[image]

Takto je to v PND
[image]

ondrejvysek, 2024-05-07:
Super, ja to mam prave 1:1, předělám to v návodu, díky.

s těmi entitami v intervalu o problému vím a zkoumám (oba soubory jsou A- Výroba), prozatím nevím přesně proč se to někde chová takto - prosím o trpělivost

Richmund7, 2024-05-07:
Kazdapodne diky za podporu a za skvely projekt!

Richmund7, 2024-05-07:
Pripadne jeste jeden podnet na vylepseni:
[image]

ondrejvysek, 2024-05-08:
je nová verze skriptu, prosím zkusit pak tu. Pozor, v konfiguraci **pnd.yaml** je nutné vyměnit **EAN** za **ELM**, aby nebylo zavádějící (viz dokumentace)

Richmund7, 2024-05-08:
Nova verze funguje, Total interval consumption se stale rovna production.

Richmund7, 2024-05-09:
Jeste jeden poznatek - v navodu doporucujete vycitat 0:30 z PND. Mne tento cas neda kompletni data za predchozi den. Ani 1:00. Az pozdeji.

ondrejvysek, 2024-05-09:
Je tam nová verze skriptu 0.9.4.4 prosím zkusit tuto. Pokud se výsledek bude tvářit stejně, prosím o nasdílení debug.zip z /pnd adresáře, viz [řešení problémů](https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND?tab=readme-ov-file#%C5%99e%C5%A1en%C3%AD-probl%C3%A9m%C5%AF-se-skriptem). Pokud bude ok, prosím také o info. Díky

Richmund7, 2024-05-09:
Prikladam debug.zip. 
[debug.zip](https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/files/15260999/debug.zip)

ondrejvysek, 2024-05-09:
Je možné v tom portále odtranit ta další okna aby tam zůstalo jenom to jediné nahoře (viz obr)? Nebo jsou k něčemu potřeba? 

[image]

Richmund7, 2024-05-09:
Okna odstranena. Problem vyresen! Diky

ondrejvysek, 2024-05-09:
Super, tak jdu ďoubat jak to zprovoznit s více okny :D

ondrejvysek, 2024-05-09:
@Richmund7  kdyžtak oprava ve verzi 0.9.4.5, více oken by již mělo být podporováno

closed 2024-05-09 by ondrejvysek