ondrejvysek/HomeAssistant-CEZDistribuce-PND #4: Senzory v HA nemá unikátní ID a nejde jej spravovat

ondrejvysek, 2024-05-08:
This entity ('sensor.pnd_consumption') does not have a unique ID, therefore its settings cannot be managed from the UI. See the [documentation](https://www.home-assistant.io/faq/unique_id) for more detail.

[image]

ondrejvysek, 2024-05-09:
Vytvořit senzory pomocí template:
```
template:
  - sensor:
      - name: "PND Data"
        icon: mdi:database-arrow-down
        unique_id: 43d1efa0-c941-4d38-930a-5dc7ce4cfd3d
        state: "{{ 'unknown' }}"
      - name: "PND Production2Consumption"
        unique_id: aeb9ef55-4db0-4ac6-93e8-d68f0ccd300d
        unit_of_measurement: "%"
        state: "{{ 'unknown' }}"
        icon: mdi:home-battery-outline
      - name: "PND Consumption"
        unique_id: 14232aec-5dd1-4ee7-92ea-6412c156c7dc
        unit_of_measurement: "kWh"
        device_class: "energy"
        icon: mdi:lightning-bolt
        state: "{{ 'unknown' }}"
      - name: "PND Production"
        unique_id: 6dc5fec0-f3c3-426b-85d4-71d2945d8109
        unit_of_measurement: "kWh"
        device_class: "energy"
        icon: mdi:lightning-bolt
        state: "{{ 'unknown' }}"
      - name: "PND Total Interval Consumption"
        unique_id: dbd8815a-480d-4be5-9348-26e848fd8013
        unit_of_measurement: "kWh"
        device_class: "energy"
        icon: mdi:lightning-bolt
        state: "{{ 'unknown' }}"
      - name: "PND Total Interval Production"
        unique_id: ba4361c6-fea7-420c-9750-65225a0a189e
        unit_of_measurement: "kWh"
        device_class: "energy"
        icon: mdi:lightning-bolt
        state: "{{ 'unknown' }}"
  - binary_sensor:
      - name: "PND Running"
        unique_id: 1f1c154f-c5ea-4eae-94c0-69425334b48b
        state: "off"
```

ondrejvysek, 2024-05-20:
přidán ukázkový [sensors.yaml](https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/blob/main/sensors.yaml), který zajistí, že senzory jsou spravovatelné v HA

closed 2024-05-20 by ondrejvysek

matomatusov, 2024-05-20:
@ondrejvysek 
Ahoj, ako správne vložiť ten binárný sensor do yaml?
[image]

Dik moc za pomoc

ondrejvysek, 2024-05-20:
Ahoj @matomatusov   u toho posledního řádku máš navíc odsazení, tj smazat ty 2 mezery před state aby bylo na úrovni unique_id a mělo by být ok.

matomatusov, 2024-05-21:
Ahoj @ondrejvysek 
keď som to opravyl tak zas nejde reštertovať HA.
[image]

Asi chýba platfom: neco

ondrejvysek, 2024-05-21:
bohužel nevidím co je komplet obsahem souborů, to co to píše, chybí `template:` někde na začátku. Pokud se nemýlím, binary_sensor používá stejnou template jako sensor, tak buď přihodit do souboru, co je template: !include nebo do template přidat více souborů:
```
# configuration.yaml
template:
  - !include binary_sensors.yaml
  - !include other_templates.yaml
```
nebo změnit strukturu souboru, aby odpovídala binary_sensor template a pak je možné ji přiložit pomocí `binary_sensor: !include binary_sensor.yaml`:
```
- platform: template
  sensors:
    pnd_runnnig:
      friendly_name: "PND Running"
      unique_id: 1f1c154f-c5ea-4eae-94c0-69425334b48b
      value_template: "{{ is_state('sensor.pnd_running', 'on') }}"
```

Zatím jsem konfiguraci sensorů řešil pouze okrajově, není to nutné pro správný běh skriptu

matomatusov, 2024-05-21:
Ďkujem moc pomohlo. Možná ešte dotaz prečo je tam to 'on' ?
 value_template: "{{ is_state('sensor.pnd_running', 'on') }}"
--------------------------------------------------------------------------

 Vo vzore je  state: "off"
  - binary_sensor:
      - name: "PND Running"
        unique_id: 1f1c154f-c5ea-4eae-94c0-69425334b48b
        state: "off"

ondrejvysek, 2024-05-21:
Je to víceméně jedno, jde o nastavení hodnoty při startu HA, podmínka říká pokud jsi on, tak dej on