ondrejvysek/HomeAssistant-CEZDistribuce-PND #5: KeyError: EAN

maxipes99, 2024-05-08:
Zdravím, mám problém, že skript spadne na KeyError EAN.

`s6-rc: info: service legacy-services successfully started
2024-05-08 15:24:15.290198 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-05-08 15:24:15.291278 INFO AppDaemon: Python version is 3.11.9
2024-05-08 15:24:15.291826 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-05-08 15:24:15.292251 INFO AppDaemon: Added log: AppDaemon
2024-05-08 15:24:15.292976 INFO AppDaemon: Added log: Error
2024-05-08 15:24:15.293302 INFO AppDaemon: Added log: Access
2024-05-08 15:24:15.293808 INFO AppDaemon: Added log: Diag
2024-05-08 15:24:15.294206 INFO AppDaemon: Added log: pnd
2024-05-08 15:24:15.632378 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-05-08 15:24:16.192565 INFO HASS: HASS Plugin Initializing
2024-05-08 15:24:16.193184 INFO HASS: HASS Plugin initialization complete
2024-05-08 15:24:16.194420 INFO AppDaemon: Initializing HTTP
2024-05-08 15:24:16.195967 INFO AppDaemon: Using 'ws' for event stream
2024-05-08 15:24:16.203373 INFO AppDaemon: Starting API
2024-05-08 15:24:16.208810 INFO AppDaemon: Starting Admin Interface
2024-05-08 15:24:16.209782 INFO AppDaemon: Starting Dashboards
2024-05-08 15:24:16.225621 INFO HASS: Connected to Home Assistant 2024.5.2
2024-05-08 15:24:16.246362 INFO AppDaemon: App 'pnd' added
2024-05-08 15:24:16.248055 INFO AppDaemon: Found 1 active apps
2024-05-08 15:24:16.248565 INFO AppDaemon: Found 0 inactive apps
2024-05-08 15:24:16.248887 INFO AppDaemon: Found 0 global libraries
2024-05-08 15:24:16.249266 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-05-08 15:24:16.250707 INFO AppDaemon: Running on port 5050
2024-05-08 15:24:16.309792 INFO HASS: Evaluating startup conditions
2024-05-08 15:24:16.315773 INFO HASS: Startup condition met: hass state=RUNNING
2024-05-08 15:24:16.316659 INFO HASS: All startup conditions met
2024-05-08 15:24:16.385811 INFO AppDaemon: Got initial state from namespace default
2024-05-08 15:24:18.265146 INFO AppDaemon: Scheduler running in realtime
2024-05-08 15:24:18.273578 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2024-05-08 15:24:18.274458 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pnd to module import path
2024-05-08 15:24:18.279192 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pnd.py
2024-05-08 15:24:20.034406 INFO AppDaemon: Loading app pnd using class pnd from module pnd
2024-05-08 15:24:20.037870 INFO AppDaemon: Calling initialize() for pnd
2024-05-08 15:24:20.216712 WARNING pnd: ------------------------------------------------------------
2024-05-08 15:24:20.217807 WARNING pnd: Unexpected error running initialize() for pnd
2024-05-08 15:24:20.218275 WARNING pnd: ------------------------------------------------------------
2024-05-08 15:24:20.221918 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pnd.py", line 45, in initialize
    self.EAN = self.args["EAN"]
               ~~~~~~~~~^^^^^^^
KeyError: 'EAN'

2024-05-08 15:24:20.222491 WARNING pnd: ------------------------------------------------------------
2024-05-08 15:24:20.224373 INFO AppDaemon: App initialization complete`

[image]

Opravdu nevím co mám špatně? Nebyla by nějaká rada. Díky moc.

ondrejvysek, 2024-05-08:
Prosím použít novou verzi skriptu 0.9.4 a vyšší, došlo k přejmenování EAN na ELM aby nedocházelo k záměně

maxipes99, 2024-05-08:
Vyřešeno po aktualizaci na verzi = "0.9.4.1". Díky moc. Skvělá práce.

ondrejvysek, 2024-05-08:
parada

closed 2024-05-08 by ondrejvysek