ondrejvysek/HomeAssistant-CEZDistribuce-PND #9: Po reštartu HA nedostupne dáta z PND

matomatusov, 2024-05-09:
Bolo by možné aby po reštartu HA sa načitaly entity z PND? Musím ručne pustiť automatizáciu aby sa to znova načítalo.

ondrejvysek, 2024-05-09:
Kouknu na možnosti. Díky tomu, že jsou senzory programově vytvářené, má limity. Jako dočasné řešení je další automatizace s trigerem startu HA a v příkladu níž je minutové čekání před spuštěním (ev se dá upravit stávající automatizace o další trigger)
```
- id: 'startup_delayed_action'
  alias: "Delayed Action After Startup"
  description: 'Performs an action one minute after Home Assistant starts.'
  trigger:
    - platform: homeassistant
      event: start
  action:
    - delay: '00:01:00'  # Delay for one minute
    - event: run_pnd
      event_data: {}
```

matomatusov, 2024-05-09:
Myslel som, že by to mohlo byť aj súčasťou tej automatizace. Dik za informace. A Super projekt!!!
Možná ešte ten súbor pnd.py by sa mohol aktualizovať cez HACS.

ondrejvysek, 2024-05-09:
ona ta upravená by mohla vypadat takto nějak:
```
alias: Run PND
description: ""
trigger:
  - platform: time
    at: "00:35:00"
  - platform: homeassistant
    event: start
condition: []
action:
  - delay:
      hours: 0
      minutes: 3
      seconds: 0
      milliseconds: 0
  - event: run_pnd
mode: single
```
nicméně ten delay ovlivní i ruční spuštění, vždy se na něj bude muset čekat. Také nevím (záleží na HW) jak dlouho startuje AppDaemon,... Asi bych spíš udělal druhou automatizaci a dal tam delay třeba 5 min. Zatím to do návodu nebudu dávat, uvidíme co se povede s těmi senzory.

ad HACS: ok, výzva přijata, pro verzi 2 nastuduji jak vytvořit HACS :) Spíš jsem potom laboroval s vlastním kontejnerem, aby nebylo závislé na AppDaemon

matomatusov, 2024-05-09:
Ono ešte dáta do PND sa aktualizujú každých 8 hodín. takže 3 x denne. Možné by to mohlo byť v tej automatizácii tiež.

ondrejvysek, 2024-05-09:
tak teoreticky lze získat i 15ti minutové vzorky, záleží co s těmi daty, aby to pak nezabilo HA. Zatím je to koncipované jako denní statistiky. Nápady super, jen tak dál - pokud bude nějaký větší zájem, můžeme prodiskutovat. Ono do toho řízne s novou legislativou povinný výkupní EAN, tak těžko říct, co to s portálem udělá.  Nejbližší plány - dostat co nejvíc zpětné vazby (funguje/nefunguje a co ne), dočistit skript a mít verzi 1, která bude funkční :)

matomatusov, 2024-05-09:
To jasné mať verzii 1 ktorá bude funční a námety na to sú. S tými dvoma EAN predsa PND funguje. Mám to od začiatku.

ondrejvysek, 2024-05-09:
To ano, ale těžko říci, jak tuto změnu promítnou do portálu,... a osobně nevím jak to tam přesně vypadá (struktura aplikace/dat), mám jenom 1 EAN. Pokud interně mají 2 EAN a na venek jedno číslo ELM, pak by to mělo fachat bezezměny

matomatusov, 2024-05-09:
Tak, že tam bude len jeden ELM ktorý má 2 EAN.

ondrejvysek, 2024-05-09:
uff, tak pak bude fungovat bezezměny, prozíravě jsem to nyní dělal na ELM a ne EAN :)

matomatusov, 2024-05-09:
Možem ešte dotaz k sensor.pnd_running . On by sa mal spustiť pri automatizácii a sám vypnuť po dokončení stažení dat?

ondrejvysek, 2024-05-09:
nikoliv při spuštění automatizace, ale při spuštění a dokončení skriptu je to jedna z prvních resp posledních akcí. Lze použít např v automatizaci, když je to po méně jak minutě, spusť skript znovu, protože portál třeba vrátil nesmysl.

gbazala, 2024-11-13:
> Bolo by možné aby po reštartu HA sa načitaly entity z PND? Musím ručne pustiť automatizáciu aby sa to znova načítalo.

Vím, že je to starší téma, možná tedy tip pro někoho, kdo to bude číst později.
Zároveň tímto ještě jednou děkuji @ondrejvysek za pomoc s instalací!

Na konci automatizace stažení dat z PND si ukládám hodnoty sensor.pnd_total_interval_consumption a  sensor.pnd_total_interval_production do input_data proměnné. Další hodnotu, kterou si ukládám do utility_meter je ze střídače denní produkce a spotřeba. Tento utility_meter resetuji také z automatizace stažení PND, ale jen pokud stažení skončí úspěšně. Do dalších výpočtů mi pak vstupují tyto hodnoty místo původních sensor.pnd_....

Možná by to šlo udělat i jednodušeji, já s HA zatím začínám...

Výhody:
- po restartu mám vždy k dispozici data
- při výpadku stahování dat z PND mám k dispozici poslední stažená data spotřeba/produkce doplněná o denní bilanci = aktuální stav
- data se mi přepočítávají průběžně bez ohledu na 8 hodinový cyklus PND
- rozdíl měření PND a střídače za 1 den není nijak tragický a při dalším stažení dat se vše srovná.

Nevýhody: 
Zaznamenávám jen celkový stav spotřeba/produkce, nemám tedy k dispozici grafy.