ondrejvysek/HomeAssistant-CEZDistribuce-PND #16: Error při spuštění skriptu

ara686, 2024-05-14:
Při spuštění skriptu do logu vypíše tuto chybu, nastavené to snad mám správně.

```
2024-05-14 15:41:26: >>>>>>>>>>>> PND Initialize
2024-05-14 15:41:26.489451 INFO AppDaemon: App initialization complete
2024-05-14 15:47:38: ********************* Starting 0.9.4.6 *********************
2024-05-14 15:47:38.522945 INFO AppDaemon: pnd: Entity binary_sensor.pnd_running created in namespace: default
2024-05-14 15:47:38: ----------------------------------------------
2024-05-14 15:47:38: Hello from AppDaemon for Portal Namerenych Dat
2024-05-14 15:47:39: Driver Loaded
2024-05-14 15:47:42: Website Opened
2024-05-14 15:48:13.295468 WARNING pnd: ------------------------------------------------------------
2024-05-14 15:48:13.297271 WARNING pnd: Unexpected error in worker for App pnd:
2024-05-14 15:48:13.298336 WARNING pnd: Worker Ags: {'id': '04346878b25d4d37a756e8212782177e', 'name': 'pnd', 'objectid': '24f2d25b90ce4c74bed739f8e63f8ccc', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f8dd65850>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2024-05-14T13:47:38.470036+00:00', 'context': {'id': '01....BM', 'parent_id': '01....MG', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2024-05-14 15:48:13.299015 WARNING pnd: ------------------------------------------------------------
2024-05-14 15:48:13.306364 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/homeassistant/appdaemon/apps/pnd.py", line 154, in run_pnd
    h1_element = wait.until(EC.presence_of_element_located((By.XPATH, f"//h1[contains(text(), '{h1_text}')]")))
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/support/wait.py", line 105, in until
    raise TimeoutException(message, screen, stacktrace)
selenium.common.exceptions.TimeoutException: Message: 

2024-05-14 15:48:13.307005 WARNING pnd: ------------------------------------------------------------
2024-05-14 15:48:13.387247 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 35.38377 seconds (limit=10)
```

ondrejvysek, 2024-05-14:
Nastaveni se zda byt ok. Zkuste pustit jeste jednou, tohle je chyba hned na zacatku scriptu po otevreni portalu (nenaslo to header). Pokud se bude vyskytovat nadale bude to na troubleshooting, treba pres FB, WA…

ara686, 2024-05-15:
To jsem zkoušel několikrát a na portále jsem byl v tu dobu přihlášený, pro jistotu jsme to zkusil ještě teď ráno, bohužel stejný problém.

ondrejvysek, 2024-05-15:
Napiste mi na ondrej@vysek.net primo, najdem kde je problem

ondrejvysek, 2024-05-19:
do verze 9.5 bude přidána kontrola přihlášení a ošetřen stav při zadání chybného hesla či jiné chybě přihlašování

closed 2024-05-19 by ondrejvysek