ondrejvysek/HomeAssistant-CEZDistribuce-PND #17: Čas pro spuštění scriptu v automatizaci

mamo-nick, 2024-05-17:
Ahoj, není to issue, spíše doporučení pro nastavení času v automatizaci.
Data do PND se načítají dvakrát denně, a to po 3.15 a 15.15.
Pokud bude čas nastaven po půlnoci, data za minulý den budou jen do 15:15. 
Doporučuji čas např. 03:30

ondrejvysek, 2024-05-19:
Ahoj, díky za comment. Dle zkušenosti se data aktualizují 3x denně po 8mi hodinách. Pokud se to chová někde trohu jinak, pustím si tu více instancí a data porovnám. Možná je to k porovnání individuálních dat - to co se stáhne a to co je ráno vidět v PND "Data včera" - pokud to odpovídá, je čas aktualizace ok. Pokud neodpovídá, pak je potřeba aktualizaci posunout.

HellixFix, 2026-09-04:
neustále narážím na to, že se data za včerejšek mění
je někde popsán aktuální stav v PND? jak se data doplňují/upřesňují? našel jsem jenom sporné/nekonkrétní info
v range csv je OK nebo N/A
N/A znamená, že se data budou ještě upřesňovat?
nebylo by vhodné doplnit nějaký senzor, který bude sledovat stav tohoto atributu třeba za včerejšek?
pak by šlo snadno rozhodnout, jestli má smysl dnes opakovaně stahovat data...

"01.09.2026 24:00:00";31.652;"naměřená data OK";
"02.09.2026 24:00:00";38.522;"naměřená data OK";
"03.09.2026 24:00:00";0.146;"N/A";
"04.09.2026 24:00:00";0;"neznámá hodnota";
"05.09.2026 24:00:00";0;"neznámá hodnota";
;