ondrejvysek/HomeAssistant-CEZDistribuce-PND #19: sensors.yaml

Richmund7, 2024-05-21:
Kam prosim nahrat ten sensors.yaml?

ondrejvysek, 2024-05-21:
Je potřeba vědět co dělám v HA, ale jsou 2 možnosti, v závislosti jak moc HA používáte, zdali již nějaké vlastní sensory jsou v konfiguraci:
1. možnost snadná
vložit obsah souboru do /homeassistant/configuration.yaml někam nakonec 
[image]

3. možnost složitější, ale do budoucna udržitelnější
- v /homeassistant/configuration.yaml přidat řádek pod ostatní !include `sensor: !include sensors.yaml`
[image]
a pak soubor sensors.yaml nahrát do /homeassistant/ (tam co je configuration.yaml)

**POZOR**: pokud již máte někde soubor sensors.yaml či jiný soubor definovaný  `sensor: !include sensors.yaml`, obsah souboru resp. pouze definice jednotlivých senzorů vložit do tohoto souboru, jinak **si zničíte již definované senzory**!

v obou případech nezapomenout na restart HomeAssitanta

Petroni2022, 2026-01-07:
Zdravím, já využil třetí možnost:
V configuration.yaml nám:

homeassistant:
  packages: !include_dir_named packages

A do /homeassistant/config/packages jsem nakopíroval přejmenovaný na pnd_sensors.yaml