ondrejvysek/HomeAssistant-CEZDistribuce-PND #21: Chyba přihlášení - tabulka_dat_button.click()

LewSY, 2024-05-22:
2024-05-22 09:29:14: ********************* Starting 0.9.5 *********************
2024-05-22 09:29:14.920091 INFO AppDaemon: pnd: Entity binary_sensor.pnd_running created in namespace: default
2024-05-22 09:29:14.931948 INFO AppDaemon: pnd: Entity sensor.pnd_script_status created in namespace: default
2024-05-22 09:29:14: ----------------------------------------------
2024-05-22 09:29:14: Hello from AppDaemon for Portal Namerenych Dat
2024-05-22 09:29:30: Driver Loaded
2024-05-22 09:29:33: Website Opened
2024-05-22 09:29:45.944077 INFO AppDaemon: pnd: Entity sensor.pnd_app_version created in namespace: default
2024-05-22 09:29:46: App Version: 0.9.5
2024-05-22 09:29:47: H1 tag with text 'Naměřená data' is present.
2024-05-22 09:29:49.042864 WARNING pnd: ------------------------------------------------------------
2024-05-22 09:29:49.043512 WARNING pnd: Unexpected error in worker for App pnd:
2024-05-22 09:29:49.043787 WARNING pnd: Worker Ags: {'id': '98e6a3b6c36b49338369a8c46458b376', 'name': 'pnd', 'objectid': 'f09e3b00cd0f45de85a9578967b3c582', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7fe8fa010f90>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2024-05-22T07:29:14.891927+00:00', 'context': {'id': '01HYFK8J4BC9DD7PYHPBPJZ43S', 'parent_id': '01HYFK8J4BM8Z2QN9Y09MFAXF6', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2024-05-22 09:29:49.044092 WARNING pnd: ------------------------------------------------------------
2024-05-22 09:29:49.070399 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/homeassistant/appdaemon/apps/pnd.py", line 222, in run_pnd
    tabulka_dat_button.click()
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/webelement.py", line 94, in click
    self._execute(Command.CLICK_ELEMENT)
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/webelement.py", line 395, in _execute
    return self._parent.execute(command, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/webdriver.py", line 347, in execute
    self.error_handler.check_response(response)
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/errorhandler.py", line 229, in check_response
    raise exception_class(message, screen, stacktrace)
selenium.common.exceptions.ElementClickInterceptedException: Message: element click intercepted: Element <button title="Export" type="button" aria-pressed="true" autocomplete="off" class="btn pnd-btn btn-outline-secondary active">...</button> is not clickable at point (343, 266). Other element would receive the click: <div id="__BVID__34" role="dialog" aria-describedby="__BVID__34___BV_modal_body_" class="modal fade show" aria-modal="true" style="display: block;">...</div>
  (Session info: chrome-headless-shell=124.0.6367.78)

2024-05-22 09:29:49.071285 WARNING pnd: ------------------------------------------------------------
2024-05-22 09:29:49.206611 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 35.206239 seconds (limit=10)
[image]

ondrejvysek, 2024-05-22:
Mám (měl jsem) také, bohužel se zobrazí pouze 1x pro přihlášeného uživatele, nedokážu se už dostat ke kódu, aby se odkliknulo. Nicméně se stačí kdekoliv přihlásit, tohe odkliknout a při dalších přihlášeních na jiných zařízeních (tam kde běží skript) se již nezobrazí a skript projde ok. Mohu zkusit nasimulovat víc, ale nemám jak vyzkoušet, díky tomu, že se nezobrazuje opakovaně :(

LewSY, 2024-05-22:
Děkuji vyzkouším ;-)

closed 2024-05-27 by ondrejvysek