ondrejvysek/HomeAssistant-CEZDistribuce-PND #22: Rozdil v datech PND vs. HA

Richmund7, 2024-05-23:
Dobry den, vsiml jsem si maleho rozdilu v datech na portalu PND a HA. 
[image]
Obdobi jsou identicky nastavene a HA resfreshnuty. Prikladam debug.zip
[debug (2).zip](https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/files/15418020/debug.2.zip)
V souborech range-consumption.csv a range-production.csv je videt, ze data zacinaji az 3.4. (obdobi mam nastavene od 1.4.) a obsahuji data i za den 24.5. (co je zitra).
Kde muzu mit chybu?
Dekuji

ondrejvysek, 2024-05-23:
Dobrý den, zkuste data z portálu exportovat ručně, co vám to v CSV ukáže (mělo by to samé, co je v CSV souborech v HA). Pokud bude ruční export 1:1 s tím co je v HA, pak je nesoulad na straně portálu. Zkusmo jsem zkusil nyní export, v období mám také zítřek, ale sestava dnes (23.5) obsahuje onu správnou hodnotu. Ten skripr s daty nedělá nic, jenom je dostane z portálu a zobrazí (vyjma výpočtu poměru mezi výrobou/spotřebou)
[image].

zkusím jim tam napsat, jak to správně interpretovat

ondrejvysek, 2024-05-23:
Koukal jsem ještě do portálu a nějaké rozdíly tam vidím, pokud se dá v PND tabulkový výpis, je to správně, ALE v tabulce je datum a **24:00**, kdežto v exportu je datum a **0:00**. Komunikuji s podporou, zde je v tom skript nevinně a nelze napravit.

[image]

Richmund7, 2024-05-23:
Tak uz to vidim. Tyto sestavy po 15 min poskytnu spravne data:
[image]
Tyto sestavy akumulujuce data po dnech uz zacnou se spozdenim dvou dnu. 
[image]
Problem 0:00 a 24:00 resim pocatkem obdobi v 0:01, jak je videt na snipu vyse.

Richmund7, 2024-05-23:
Asi to mam vyresene. Kdyz posunu v dennich sestavach zacatek o den driv dostanu spravne udaje. Data s zitrejsim datem 24/05/2024  00:00:00 jsou vlastne za dnesek. Takze jen staci upravit rozpeti obdobi s vedomim tohoto posunu.

ondrejvysek, 2024-05-24:
Komunikace z ČEZ Distribuce:
**Dotaz:**
Správnost dat – porovnával jsem data (23.5.2024 16:50) A+ (stejná situace je i u A-) a nesouhlasí správnost pokud zvolím vlastní období tak se data liší:
a.	V tabulkovém výpisu je vše správně, stejně tak když se zvolí období „dnes“
b.	V exportu (XLSX / CSV) se poslední datum ukazuje v následujícím dni – viz obrázek níže. Jak prosím zajistit, aby oba výstupy byly identické? Všiml jsem si rozdílu kdy v tabulce je datum a 24:00, kdežto v exportu je datum a 0:00. nicméně dnešní data nemohou být zítra 😊
 (obrázek stejný jako výše)

 **Odpověď:**
Bohužel, o této chybě víme a dodavatel pracuje na odstranění. V případě exportování dat ve 1/4h profilu k problému nedochází. Za komplikace se omlouváme.

**Komentář:**
Skript na 1/4 vzorky předělávat nebudu, významně by přibylo dat v HA. Ona se ta dlouhodobá statistika postupně udělá z denních senzorů, které jsou jako energy a tedy uchovávané trvale.

closed 2024-05-24 by ondrejvysek