ondrejvysek/HomeAssistant-CEZDistribuce-PND #24: AppDaemon  No app description found for: /homeassistant/appdaemon/apps/pnd.py - ignoring

kyselyna, 2024-05-27:
Po spuštění automatizace se nic neděje neobjeví se žádné entity..

[image]
[image]
[image]

ondrejvysek, 2024-05-27:
Jak prosím vypadá ta automatizace? můžete v DevTools zkusit událost vyvolat ručně?
- Dev Tools > Events > Event type: **run_pnd** > Fire Event
[image]

ondrejvysek, 2024-05-27:
můžete prosím kouknout, zdali máte tento soubor /homeassistant/appdaemon.yaml? Pokud ano, můžete zkusit appdaemon.yaml konfiguraci dát i do něj? pak restartovat appdaemon
[image]

kyselyna, 2024-05-27:
> Jak prosím vypadá ta automatizace? můžete v DevTools zkusit událost vyvolat ručně?
> 
> * Dev Tools > Events > Event type: **run_pnd** > Fire Event
>   ![obraz](https://private-user-images.githubusercontent.com/29834082/334010008-192c42e5-92c8-4afd-9003-9d152278e950.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY3OTY0MTQsIm5iZiI6MTcxNjc5NjExNCwicGF0aCI6Ii8yOTgzNDA4Mi8zMzQwMTAwMDgtMTkyYzQyZTUtOTJjOC00YWZkLTkwMDMtOWQxNTIyNzhlOTUwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI3VDA3NDgzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4MTExM2M2MDQyYjY4ZGMzZDVjZDMxMjY0NDg1ZTQ4ZWU0MDMxMjliOTU3MDVmYTgwNjliZjJhYmMwYWI2YzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IZQJq8rohp5NW75jBtAqIPgk6bgQBZAa5kjTsU8tBgw)

[image]

ondrejvysek, 2024-05-27:
Zkusil bych. Log píše odkud bere konfiguraci, pokud tam není, tak možná hledá marně.

kyselyna, 2024-05-27:
zduplikoval jsem soubor z addon_configs....  ale furt nemám žádné soubory v /homeassistant/appdaemon/apps/pnd  níže posílám log:

Executing busybox-1.36.1-r15.trigger
Executing fontconfig-2.14.2-r4.trigger
Executing mkfontscale-1.2.2-r4.trigger
Executing glib-2.78.6-r0.trigger
Executing shared-mime-info-2.4-r0.trigger
Executing gdk-pixbuf-2.42.10-r6.trigger
Executing gtk-update-icon-cache-3.24.41-r0.trigger
Executing gtk+3.0-3.24.41-r0.trigger
OK: 676 MiB in 208 packages
OK: 676 MiB in 208 packages
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting selenium
  Downloading https://wheels.home-assistant.io/musllinux-index/selenium-4.21.0-py3-none-any.whl (9.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 9.5/9.5 MB 4.9 MB/s eta 0:00:00
Requirement already satisfied: urllib3<3,>=1.26 in /usr/lib/python3.11/site-packages (from urllib3[socks]<3,>=1.26->selenium) (1.26.18)
Collecting trio~=0.17 (from selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/trio-0.25.1-py3-none-any.whl (467 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 467.7/467.7 kB 5.8 MB/s eta 0:00:00
Collecting trio-websocket~=0.9 (from selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/trio_websocket-0.11.1-py3-none-any.whl (17 kB)
Requirement already satisfied: certifi>=2021.10.8 in /usr/lib/python3.11/site-packages (from selenium) (2024.2.2)
Collecting typing_extensions>=4.9.0 (from selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/typing_extensions-4.12.0-py3-none-any.whl (37 kB)
Requirement already satisfied: attrs>=23.2.0 in /usr/lib/python3.11/site-packages (from trio~=0.17->selenium) (23.2.0)
Collecting sortedcontainers (from trio~=0.17->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/sortedcontainers-2.4.0-py2.py3-none-any.whl.metadata (10 kB)
Requirement already satisfied: idna in /usr/lib/python3.11/site-packages (from trio~=0.17->selenium) (3.7)
Collecting outcome (from trio~=0.17->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/outcome-1.3.0.post0-py2.py3-none-any.whl (10 kB)
Collecting sniffio>=1.3.0 (from trio~=0.17->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/sniffio-1.3.1-py3-none-any.whl (10 kB)
Requirement already satisfied: wsproto>=0.14 in /usr/lib/python3.11/site-packages (from trio-websocket~=0.9->selenium) (1.2.0)
Collecting PySocks!=1.5.7,<2.0,>=1.5.6 (from urllib3[socks]<3,>=1.26->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/PySocks-1.7.1-py3-none-any.whl (16 kB)
Requirement already satisfied: h11<1,>=0.9.0 in /usr/lib/python3.11/site-packages (from wsproto>=0.14->trio-websocket~=0.9->selenium) (0.14.0)
Downloading https://wheels.home-assistant.io/musllinux-index/sortedcontainers-2.4.0-py2.py3-none-any.whl (29 kB)
Installing collected packages: sortedcontainers, typing_extensions, sniffio, PySocks, outcome, trio, trio-websocket, selenium
Successfully installed PySocks-1.7.1 outcome-1.3.0.post0 selenium-4.21.0 sniffio-1.3.1 sortedcontainers-2.4.0 trio-0.25.1 trio-websocket-0.11.1 typing_extensions-4.12.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting pandas
  Downloading pandas-2.2.2-cp311-cp311-musllinux_1_1_aarch64.whl.metadata (19 kB)
Collecting numpy>=1.23.2 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/numpy-1.26.4-cp311-cp311-musllinux_1_2_aarch64.whl (11.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.3/11.3 MB 5.2 MB/s eta 0:00:00
Requirement already satisfied: python-dateutil>=2.8.2 in /usr/lib/python3.11/site-packages (from pandas) (2.8.2)
Requirement already satisfied: pytz>=2020.1 in /usr/lib/python3.11/site-packages (from pandas) (2023.3.post1)
Collecting tzdata>=2022.7 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/tzdata-2024.1-py2.py3-none-any.whl (345 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 345.4/345.4 kB 5.8 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3.11/site-packages (from python-dateutil>=2.8.2->pandas) (1.16.0)
Downloading pandas-2.2.2-cp311-cp311-musllinux_1_1_aarch64.whl (16.3 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.3/16.3 MB 5.2 MB/s eta 0:00:00
Installing collected packages: tzdata, numpy, pandas
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Successfully installed numpy-1.26.4 pandas-2.2.2 tzdata-2024.1
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting bs4
  Downloading https://wheels.home-assistant.io/musllinux-index/bs4-0.0.2-py2.py3-none-any.whl.metadata (411 bytes)
Collecting beautifulsoup4 (from bs4)
  Downloading https://wheels.home-assistant.io/musllinux-index/beautifulsoup4-4.12.3-py3-none-any.whl (147 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 147.9/147.9 kB 567.3 kB/s eta 0:00:00
Collecting soupsieve>1.2 (from beautifulsoup4->bs4)
[... cut ...]
2024-05-27 10:34:31.681452 INFO HASS: All startup conditions met
2024-05-27 10:34:31.754295 INFO AppDaemon: Got initial state from namespace default
2024-05-27 10:34:33.629195 INFO AppDaemon: Scheduler running in realtime
2024-05-27 10:34:33.634338 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2024-05-27 10:34:33.635468 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pnd to module import path
2024-05-27 10:34:33.636340 INFO AppDaemon: App initialization complete

ondrejvysek, 2024-05-27:
Muzete zkusit spustit pres ty dev tools?

kyselyna, 2024-05-27:
nic víc to neudělá :(

[image]

kyselyna, 2024-05-27:
[image]

ondrejvysek, 2024-05-27:
A log appdaemona nyni vypada jak? Porad stejne?

kyselyna, 2024-05-27:
> A log appdaemona nyni vypada jak? Porad stejne?

Bohužel ano ani se nehne přijde mi že vbc nereaguje na příkaz. Je jedno jestli to zkouším skrz dev tools nebo přes automatizaci. 

Níže je automatizace:
alias: Run PDN
description: ""
trigger:
  - platform: time
    at: "00:30:00"
condition: []
action:
  - event: run_pnd
    event_data: {}
mode: single

ondrejvysek, 2024-05-27:
Problém je v tom, že appdaemon nastartuje ok, ale vůbec neregistruje tu pnd aplikaci

ondrejvysek, 2024-05-27:
Na guglu se několikrát podobný problém vyskytuje (WARNING AppDaemon: No app description found for), ale s žádným rozumným řešením. Zkusil bych:
* resrtart celého HA (pokud nebylo dřív)
* v appdaemon.yaml vyhodit tento řádek: `app_dir: /homeassistant/appdaemon/apps`
* apps.yaml vytvořit zde \addon_configs\a0d7b954_appdaemon\apps
* pnd.py dát také do \addon_configs\a0d7b954_appdaemon\apps
* restart appdaemon

kdyžtak napište na ondrej@vysek.net, můžeme zkusit kouknout přes vzdálenou plochu.

ondrejvysek, 2024-05-27:
Jeste prosim zkuste nahradit obsah souboru pnd.py rucne (nestahujte soubor) ale zkopirujte kod a vlozte do ha. Dnes se objevil podobny problem a toto bylo reseni

kyselyna, 2024-05-27:
> Na guglu se několikrát podobný problém vyskytuje (WARNING AppDaemon: No app description found for), ale s žádným rozumným řešením. Zkusil bych:
> 
> * resrtart celého HA (pokud nebylo dřív)
> * v appdaemon.yaml vyhodit tento řádek: `app_dir: /homeassistant/appdaemon/apps`
> * apps.yaml vytvořit zde \addon_configs\a0d7b954_appdaemon\apps
> * pnd.py dát také do \addon_configs\a0d7b954_appdaemon\apps
> * restart appdaemon
> 
> kdyžtak napište můžeme zkusit kouknout přes vzdálenou plochu.

tak tohle pomohlo ono to furt bralo ten původní příkaz "hello word" který byl v \addon_configs\a0d7b954_appdaemon\apps

LOG:
[22:06:03] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-05-27 22:06:08.663518 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-05-27 22:06:08.664041 INFO AppDaemon: Python version is 3.11.9
2024-05-27 22:06:08.664395 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-05-27 22:06:08.664700 INFO AppDaemon: Added log: AppDaemon
2024-05-27 22:06:08.665000 INFO AppDaemon: Added log: Error
2024-05-27 22:06:08.665250 INFO AppDaemon: Added log: Access
2024-05-27 22:06:08.665525 INFO AppDaemon: Added log: Diag
2024-05-27 22:06:08.665805 INFO AppDaemon: Added log: pnd
2024-05-27 22:06:08.971687 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-05-27 22:06:09.531172 INFO HASS: HASS Plugin Initializing
2024-05-27 22:06:09.531505 INFO HASS: HASS Plugin initialization complete
2024-05-27 22:06:09.532180 INFO AppDaemon: Initializing HTTP
2024-05-27 22:06:09.533645 INFO AppDaemon: Using 'ws' for event stream
2024-05-27 22:06:09.538912 INFO AppDaemon: Starting API
2024-05-27 22:06:09.542588 INFO AppDaemon: Starting Admin Interface
2024-05-27 22:06:09.543352 INFO AppDaemon: Starting Dashboards
2024-05-27 22:06:09.559840 INFO HASS: Connected to Home Assistant 2024.5.5
2024-05-27 22:06:09.579216 INFO AppDaemon: App 'pnd' added
2024-05-27 22:06:09.581286 INFO AppDaemon: Found 1 active apps
2024-05-27 22:06:09.582035 INFO AppDaemon: Found 0 inactive apps
2024-05-27 22:06:09.582764 INFO AppDaemon: Found 0 global libraries
2024-05-27 22:06:09.583514 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-05-27 22:06:09.585462 INFO AppDaemon: Running on port 5050
2024-05-27 22:06:09.610753 INFO HASS: Evaluating startup conditions
2024-05-27 22:06:09.616514 INFO HASS: Startup condition met: hass state=RUNNING
2024-05-27 22:06:09.616939 INFO HASS: All startup conditions met
2024-05-27 22:06:09.682803 INFO AppDaemon: Got initial state from namespace default
2024-05-27 22:06:11.618243 INFO AppDaemon: Scheduler running in realtime
2024-05-27 22:06:11.623165 INFO AppDaemon: Adding /config/apps to module import path
2024-05-27 22:06:11.628255 WARNING AppDaemon: No app description found for: /config/apps/hello.py - ignoring
2024-05-27 22:06:11.630837 INFO AppDaemon: Loading App Module: /config/apps/pnd.py
2024-05-27 22:06:12.891798 INFO AppDaemon: Loading app pnd using class pnd from module pnd
2024-05-27 22:06:12.894372 INFO AppDaemon: Calling initialize() for pnd
2024-05-27 22:06:12: >>>>>>>>>>>> PND Initialize
2024-05-27 22:06:12.897665 INFO AppDaemon: App initialization complete
2024-05-27 22:06:46: ********************* Starting 0.9.5 *********************
2024-05-27 22:06:46: ----------------------------------------------
2024-05-27 22:06:46: Hello from AppDaemon for Portal Namerenych Dat
2024-05-27 22:06:47: Driver Loaded
2024-05-27 22:06:50: Website Opened
2024-05-27 22:07:03: App Version: 2.3.5
2024-05-27 22:07:06: H1 tag with text 'Naměřená data' is present.
2024-05-27 22:07:14: Rychla Sestava selected successfully!
2024-05-27 22:07:16: Selecting ELM '3000013984'
2024-05-27 22:07:16: Valid ELM numbers 'ELM 3000013984, ELM 3000013984, ELM 3000013984'
2024-05-27 22:07:20: ELM Status: ELM 3000013984 - 3000013984
2024-05-27 22:07:20: Iteration 0: Vyhledat Button NOT disabled
2024-05-27 22:07:20: Device ELM '3000013984' selected successfully!
2024-05-27 22:07:23: Button 'Vyhledat data' clicked successfully!
2024-05-27 22:07:27: 07 Profil spotřeby za den (+A)
2024-05-27 22:07:36: Downloading CSV file for 07 Profil spotřeby za den (+A)
2024-05-27 22:07:37: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-consumption.csv
2024-05-27 22:07:38: 08 Profil výroby za den (-A)
2024-05-27 22:07:46: Downloading CSV file for 08 Profil výroby za den (-A)
2024-05-27 22:07:48: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-production.csv
2024-05-27 22:07:48: All Done - DAILY DATA DOWNLOADED
2024-05-27 22:07:48: Latest entry: 27.05.2024 23:59 - 13.336 kWh
2024-05-27 22:07:48: Latest entry: 27.05.2024 23:59 - 32.365 kWh
2024-05-27 22:07:48: All Done - DAILY DATA PROCESSED
2024-05-27 22:07:51: Data Interval Entered - '01.10.2023 00:00 - 01.10.2024 00:00'

ondrejvysek, 2024-05-27:
Tak parada. Chova se to nejak zvlastne, prave ten app_dir parametr by mel zajistit ignoraci hello_world a vzit apps jinde, coz se delo jenom z casti

closed 2024-05-27 by ondrejvysek