ondrejvysek/HomeAssistant-CEZDistribuce-PND #31: Měsíční data nesouhlasí s fakturací

luboskopal, 2024-06-01:
Narazil jsem na jeden problém:
Pokud je v csv souboru "range-consumption" u denního záznamu ve sloupci Status uvedeno "naměřená data, výpadek napětí", tak v grafu s měsíční agregací je tento záznam přeskočen. Následně je do množiny zahrnut záznam z dalšího měsíce, aby souhlasil počet záznamů.
V sensoru PND_data je hodnota tohoto nekorektního záznamu.
V oficiální měsíční fakturaci je tento záznam zahrnut.

Příklad:
20.2.2024 mám v souboru range-consumption uvedeno "naměřená data, výpadek napětí". Hodnota spotřeby v záznamu je. V sensoru PND_data je také.
V grafu měsíční spotřeby je spotřeba tohoto dne vynechána a do výběru dat je přidán záznam 02.03.2024  0:00:00.

ondrejvysek, 2024-06-01:
Skript nijak nepracuje se stavy měření. Rozumím správně, že informace pnd_data = oficiální fakturace a problém je pouze s grafem? Prosím také kouknout zde #15 apex není úplně ideální na vizualizaci :(

luboskopal, 2024-06-01:
Ano, chápete to správně. Problém vidím jen v tom grafu.

ondrejvysek, 2024-06-01:
ok, pak prosím zkusit změnit na ten week jak je v #15 trochu by se mělo zlepšit, problém nevyřeší, ďoubal jsem APEX a řešení zatím nenašel. Nechám toto otevřené, zkusím kouknout zdali třeba Plotly by nebylo lepší

luboskopal, 2024-06-01:
Omlouvám se. Věc se má trochu jinak.
Překopíroval jsem data z csv souboru do Excelu a porovnal následující měsíc (březen).
Graf bere záznamy od 03.03.2024  0:00:00 do 02.04.2024  0:00:00. 
Od února je to celé posunuté. Mé původní tvrzení bylo chybné. Stav měření na to nemá vliv.

ondrejvysek, 2024-06-01:
posunutá data v CSV? koukněte na #28 - pokud je problém v podkladových datech, musíme si počkat, nežli to čez opraví 
posunutý graf? koukněte na #15 - pokud je problém ve vizualizaci, 1) Apex není úplně nej s agregacemi 2) není úplně kámoš s přestupným rokem.

zkuste trochu přiblížit co je ok a kde se problém projevuje (data v CSV, data v atributu, data v grafu), pomůže to lépe identifikovat možné řešení.

luboskopal, 2024-06-01:
V souboru range-consuption.csv jsem dohledal data použitá pro jednotlivé měsíce grafu měsíční agregace:
Leden jsou záznamy od 02.01.2024  0:00:00 do 01.02.2024  0:00:00 -  31 záznamů - součet souhlasí s fakturací
Únor jsou záznamy od 02.02.2024  0:00:00 do 02.03.2024  0:00:00 - 30 záznamů - součet nesouhlasí s fakturací
Březen jsou záznamy od 03.03.2024 0:00:00 do 02.04.2024 0:00:00 - 31 záznamů - součet nesouhlasí s fakturací
Duben jsou záznamy od 03.04.2024  0:00:00 do 02.05.2024  0:00:00 - 30 záznamů - součet nesouhlasí s fakturací

Zkoušel jsem stáhnout data za loňský rok, abych vyloučil vliv přestupného roku, ale nepovedlo se to. Asi na PNG portále už nejsou data.

Kontroloval jsem data z faktury vůči souboru CSV. Jsou v pořádku. Tam problém není.
Data v atributu jsou také v pořádku.
Vypadá to na problém s grafem.

ondrejvysek, 2024-06-01:
Díky za analýzu, pokud jsou data v CSV ok, pak budou (by měla být) i v atributu, sktipt je jenom lopata, proparsuje a hodí. 

Jak jsem psal, Apex není úplně nejlepší pro tuhle vizualizaci, viz ten problém #15. Zkusím někdy po večerech najít jinou alternativu, v Plotly, ale znamená to komplet předělat logiku zobrazení a získání dat, což je pro mne nové = experimentování :) Nicméně Plotly by mohlo nabízet i lepší filtrování,...