ondrejvysek/HomeAssistant-CEZDistribuce-PND #32: Nenčítají se hodnoty z PND

Rotif, 2024-06-01:
Ahoj,
z webu čezu se mi nenačtou žádné hodnoty.
V logu se vždy po spuštění automatizace zobrazuje:
2024-06-01 00:30:56.650143 WARNING pnd: Coroutine (<coroutine object ADAPI.set_state at 0x7f88c30a90>) took too long (10 seconds), cancelling the task...
Přitom ale v složce pnd jsou screenshoty, které zobrazují volbu elektroměru a hodnot na PND.
Co dělám špatně?
Dík

ondrejvysek, 2024-06-01:
Prosim:
1. Jakym screenshotem to konci? Jsou cislovane
2. Prosim o vic logu z appdaemon. Skript se evidentne spusti, tak v logu musi byt detail

Rotif, 2024-06-01:
Screenshoty z pnd log, víc nevím, kde bych hledal:
[image]
[image]
[image]
[image]
[image]
[debug-ELM.txt](https://github.com/user-attachments/files/15523224/debug-ELM.txt)
[pnd.log](https://github.com/user-attachments/files/15523225/pnd.log)

ondrejvysek, 2024-06-01:
Kouknu na to, logy appdaemon jsou v doplnku appdaemon, nahore je zalozka log, tak tam. Jak prosim vypada apps.yaml? Predevsim ELM

Rotif, 2024-06-02:
V příloze log a apps
[log appdaemon.txt](https://github.com/user-attachments/files/15524481/log.appdaemon.txt)

[apps_yaml.txt](https://github.com/user-attachments/files/15524484/apps_yaml.txt)
Dík

Rotif, 2024-06-02:
Omlouvám se, zadal jsem chybné číslo elektroměru, opravím to a dám vědět.

ondrejvysek, 2024-06-02:
Super, on to píše i ten log, že nenalezl ELM:
```
  File "/homeassistant/appdaemon/apps/pnd.py", line 294, in run_pnd
    raise Exception(f"Failed to find '{self.ELM}' in the selection")
Exception: Failed to find '37120345' in the selection
```

closed 2024-06-02 by ondrejvysek