ondrejvysek/HomeAssistant-CEZDistribuce-PND #36: Unexpected error loading module: /homeassistant/appdaemon/apps/pnd.py

TesaTesa, 2024-06-05:
Dobrý večer,

nedaří se mi úspěšně spustit script k vyčtení hodnost z distribučního portálu - nevytvoří se ani nové entity, ani případný log který bych mohl poslat.

mám poslední / updatovanou verzi HA běžící na RPI5.

Zde je výpis z logu Appdaemonu :

[21:13:01] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-06-05 21:13:03.701039 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-06-05 21:13:03.701208 INFO AppDaemon: Python version is 3.11.9
2024-06-05 21:13:03.701279 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-06-05 21:13:03.701341 INFO AppDaemon: Added log: AppDaemon
2024-06-05 21:13:03.701410 INFO AppDaemon: Added log: Error
2024-06-05 21:13:03.701476 INFO AppDaemon: Added log: Access
2024-06-05 21:13:03.701553 INFO AppDaemon: Added log: Diag
2024-06-05 21:13:03.701659 INFO AppDaemon: Added log: pnd
2024-06-05 21:13:03.831059 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-06-05 21:13:04.061789 INFO HASS: HASS Plugin Initializing
2024-06-05 21:13:04.061954 INFO HASS: HASS Plugin initialization complete
2024-06-05 21:13:04.062236 INFO AppDaemon: Initializing HTTP
2024-06-05 21:13:04.062751 INFO AppDaemon: Using 'ws' for event stream
2024-06-05 21:13:04.064616 INFO AppDaemon: Starting API
2024-06-05 21:13:04.065683 INFO AppDaemon: Starting Admin Interface
2024-06-05 21:13:04.065863 INFO AppDaemon: Starting Dashboards
2024-06-05 21:13:04.071596 INFO HASS: Connected to Home Assistant 2024.5.5
2024-06-05 21:13:04.079365 INFO AppDaemon: App 'pnd' added
2024-06-05 21:13:04.080007 INFO AppDaemon: Found 1 active apps
2024-06-05 21:13:04.080138 INFO AppDaemon: Found 0 inactive apps
2024-06-05 21:13:04.080220 INFO AppDaemon: Found 0 global libraries
2024-06-05 21:13:04.080335 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-06-05 21:13:04.081013 INFO AppDaemon: Running on port 5050
2024-06-05 21:13:04.092268 INFO HASS: Evaluating startup conditions
2024-06-05 21:13:04.094900 INFO HASS: Startup condition met: hass state=RUNNING
2024-06-05 21:13:04.095067 INFO HASS: All startup conditions met
2024-06-05 21:13:04.115901 INFO AppDaemon: Got initial state from namespace default
2024-06-05 21:13:06.085047 INFO AppDaemon: Scheduler running in realtime
2024-06-05 21:13:06.086903 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2024-06-05 21:13:06.087173 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pnd to module import path
2024-06-05 21:13:06.088137 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pnd.py
2024-06-05 21:13:06.104203 WARNING Error: ------------------------------------------------------------
2024-06-05 21:13:06.104428 WARNING Error: Unexpected error loading module: /homeassistant/appdaemon/apps/pnd.py:
2024-06-05 21:13:06.104694 WARNING Error: ------------------------------------------------------------
2024-06-05 21:13:06.109292 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/homeassistant/appdaemon/apps/pnd.py", line 106
    <title>HomeAssistant-CEZDistribuce-PND/pnd.py at main · ondrejvysek/HomeAssistant-CEZDistribuce-PND · GitHub</title>
                                                          ^
SyntaxError: invalid character '·' (U+00B7)

2024-06-05 21:13:06.109562 WARNING Error: ------------------------------------------------------------
2024-06-05 21:13:06.109763 WARNING AppDaemon: Removing associated apps:
2024-06-05 21:13:06.109943 WARNING AppDaemon: pnd
2024-06-05 21:13:06.110344 INFO AppDaemon: App initialization complete

Děkuji za Váš čas :)

Radek

TesaTesa, 2024-06-06:
Děkuji Vám Ondro,

jako vždy, za vším hledejme chybu koncového uživatele :D Jako PND.py jsem uložil webovou stránku, avšak ne pythonový kód ...

Po nahrazení kódem vše funguje jak má. Achjo .. :)

Děkuji Vám moc !

Radek

ondrejvysek, 2024-06-06:
Super ze se povedlo!

closed 2024-06-06 by ondrejvysek