ondrejvysek/HomeAssistant-CEZDistribuce-PND #39: Docker AppDaemon stáhne se pouze část dat

K4BlOs, 2024-06-13:
Provozuji HA na dockeru stejně tak AppDaemon. Postupoval jsem dle návodu dá se říct, že z poloviny mi váš doplňek funguje. Při spuštění scriptu run_pnd automatizace v HA se mi stáhne část dat
[image]
bohužel ale už nedojde ke stažení zip souboru. Přikládám log z AppDaemon

2024-06-13 14:44:15.477516 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-06-13 14:44:15.477605 INFO AppDaemon: Python version is 3.10.11
2024-06-13 14:44:15.477677 INFO AppDaemon: Configuration read from: /conf/appdaemon.yaml
2024-06-13 14:44:15.477744 INFO AppDaemon: Added log: AppDaemon
2024-06-13 14:44:15.477815 INFO AppDaemon: Added log: Error
2024-06-13 14:44:15.477883 INFO AppDaemon: Added log: Access
2024-06-13 14:44:15.477962 INFO AppDaemon: Added log: Diag
2024-06-13 14:44:15.478027 INFO AppDaemon: Added log: pnd
2024-06-13 14:44:15.493297 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-06-13 14:44:15.602184 INFO HASS: HASS Plugin Initializing
2024-06-13 14:44:15.602279 INFO HASS: HASS Plugin initialization complete
2024-06-13 14:44:15.602486 INFO AppDaemon: Initializing HTTP
2024-06-13 14:44:15.602797 INFO AppDaemon: Using 'ws' for event stream
2024-06-13 14:44:15.604429 INFO AppDaemon: Starting API
2024-06-13 14:44:15.605560 INFO AppDaemon: Starting Admin Interface
2024-06-13 14:44:15.605728 INFO AppDaemon: Starting Dashboards
2024-06-13 14:44:15.610197 INFO HASS: Connected to Home Assistant 2024.6.1
2024-06-13 14:44:15.615562 INFO AppDaemon: App 'pnd' added
2024-06-13 14:44:15.616112 INFO AppDaemon: Found 1 active apps
2024-06-13 14:44:15.616250 INFO AppDaemon: Found 0 inactive apps
2024-06-13 14:44:15.616376 INFO AppDaemon: Found 0 global libraries
2024-06-13 14:44:15.616505 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-06-13 14:44:15.616920 INFO AppDaemon: Running on port 5050
2024-06-13 14:44:15.623277 INFO HASS: Evaluating startup conditions
2024-06-13 14:44:15.624736 INFO HASS: Startup condition met: hass state=RUNNING
2024-06-13 14:44:15.624859 INFO HASS: All startup conditions met
2024-06-13 14:44:15.633895 INFO AppDaemon: Got initial state from namespace default
2024-06-13 14:44:17.628753 INFO AppDaemon: Scheduler running in realtime
2024-06-13 14:44:17.632666 INFO AppDaemon: Adding /conf/apps to module import path
2024-06-13 14:44:17.633365 INFO AppDaemon: Adding /conf/apps/pnd to module import path
2024-06-13 14:44:17.635993 INFO AppDaemon: Loading App Module: /conf/apps/pnd.py
2024-06-13 14:44:17.917130 INFO AppDaemon: Loading app pnd using class pnd from module pnd
2024-06-13 14:44:17.918014 INFO AppDaemon: Calling initialize() for pnd
2024-06-13 14:44:17: >>>>>>>>>>>> PND Initialize
2024-06-13 14:44:17.918967 INFO AppDaemon: App initialization complete
2024-06-13 14:44:35: ********************* Starting 0.9.7 *********************
2024-06-13 14:44:35: ----------------------------------------------
2024-06-13 14:44:35: Hello from AppDaemon for Portal Namerenych Dat
2024-06-13 14:44:36: Driver Loaded
2024-06-13 14:44:37: Website Opened
2024-06-13 14:44:47: App Version: 2.3.5
2024-06-13 14:44:47: H1 tag with text 'Naměřená data' is present.
2024-06-13 14:44:52: Rychla Sestava selected successfully!
2024-06-13 14:44:53: Selecting ELM '792648'
2024-06-13 14:44:53: Valid ELM numbers 'ELM 792648, ELM 792648, ELM 792648, ELM 792648'
2024-06-13 14:44:55: ELM Status:  - 792648
2024-06-13 14:44:55: Iteration 0: Vyhledat Button IS disabled
2024-06-13 14:44:56: ELM Status: ELM 792648 - 792648
2024-06-13 14:44:56: Iteration 1: Vyhledat Button NOT disabled
2024-06-13 14:44:56: Device ELM '792648' selected successfully!
2024-06-13 14:44:57: Button 'Vyhledat data' clicked successfully!
2024-06-13 14:45:00: 07 Profil spotřeby za den (+A)
2024-06-13 14:45:05: Downloading CSV file for 07 Profil spotřeby za den (+A)
2024-06-13 14:45:06: File downloaded and saved as: /conf/apps/pnd/daily-consumption.csv
2024-06-13 14:45:06: 08 Profil výroby za den (-A)
2024-06-13 14:45:11: Downloading CSV file for 08 Profil výroby za den (-A)
2024-06-13 14:45:12: File downloaded and saved as: /conf/apps/pnd/daily-production.csv
2024-06-13 14:45:12: All Done - DAILY DATA DOWNLOADED
2024-06-13 14:45:12.923480 WARNING pnd: ------------------------------------------------------------
2024-06-13 14:45:12.923647 WARNING pnd: Unexpected error in worker for App pnd:
2024-06-13 14:45:12.923789 WARNING pnd: Worker Ags: {'id': '82fa39d2ed414269ab0806c271d17d2f', 'name': 'pnd', 'objectid': '7e5f48e0f51043479a98c6a386d0b109', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f69c3726c80>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2024-06-13T12:44:35.763697+00:00', 'context': {'id': '01J08T1SHJMEBHWNY24750CKDS', 'parent_id': '01J08T1SHJPY7PK4B6QR2QXHED', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2024-06-13 14:45:12.923903 WARNING pnd: ------------------------------------------------------------
2024-06-13 14:45:12.925818 WARNING pnd: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/conf/apps/pnd.py", line 482, in run_pnd
    data_consumption = pd.read_csv(self.download_folder + '/daily-consumption.csv', delimiter=';', encoding='latin1')
  File "/usr/local/lib/python3.10/site-packages/pandas/io/parsers/readers.py", line 1026, in read_csv
    return _read(filepath_or_buffer, kwds)
  File "/usr/local/lib/python3.10/site-packages/pandas/io/parsers/readers.py", line 626, in _read
    return parser.read(nrows)
  File "/usr/local/lib/python3.10/site-packages/pandas/io/parsers/readers.py", line 1923, in read
    ) = self._engine.read(  # type: ignore[attr-defined]
  File "/usr/local/lib/python3.10/site-packages/pandas/io/parsers/c_parser_wrapper.py", line 234, in read
    chunks = self._reader.read_low_memory(nrows)
  File "parsers.pyx", line 838, in pandas._libs.parsers.TextReader.read_low_memory
  File "parsers.pyx", line 905, in pandas._libs.parsers.TextReader._read_rows
  File "parsers.pyx", line 874, in pandas._libs.parsers.TextReader._tokenize_rows
  File "parsers.pyx", line 891, in pandas._libs.parsers.TextReader._check_tokenize_status
  File "parsers.pyx", line 2061, in pandas._libs.parsers.raise_parser_error
pandas.errors.ParserError: Error tokenizing data. C error: Expected 1 fields in line 8, saw 2
2024-06-13 14:45:12.925955 WARNING pnd: ------------------------------------------------------------
2024-06-13 14:45:12.980293 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 37.98007 seconds (limit=10)

Cesty pro stahování jsem upravil takto:
appdaemon.yaml
`appdaemon:
  latitude: 50.1724921
  longitude: 12.3555073
  elevation: 478.2
  time_zone: Europe/Prague
  plugins:
    HASS:
      type: hass
      ha_url: http://192.168.1.4:8123/
      token: muj token
      cert_verify: False
  app_dir: /conf/apps
http:
  url: http://192.168.1.4:5050
admin:
api:
hadashboard:
logs:
  pnd:
    name: pnd
    filename: /conf/apps/pnd/pnd.log`

apps.yaml
`---
pnd:
  module: pnd
  class: pnd
  log: pnd
  PNDUserName: "muj email"
  PNDUserPassword: "moje heslo"
  DataInterval: "27.10.2023 00:00 - 27.10.2024 00:00"
  ELM: "muj ELM"
  DownloadFolder: "/conf/apps/pnd/"`

pnd.py  
Zde jsem na konci kodu nasel cesty tak jsem take upravil, zkousel jsem i cestu v dockeru tedy /conf/.... a i celou cestu, ale ani s jednou script nefunguje uplně.
`    # Close the browser
    driver.quit()
    print(dt.now().strftime("%Y-%m-%d %H:%M:%S") + ": " + "All Done - BROWSER CLOSED")
    self.set_state("binary_sensor.pnd_running", state="off")
    print(dt.now().strftime("%Y-%m-%d %H:%M:%S") + ": " + "Sensor State Set to OFF")
    zip_folder("/home/karel/Docker/appdaemon/config/apps/pnd/", "/home/karel/Docker/appdaemon/config/apps/debug.zip")
    shutil.move("/home/karel/Docker/appdaemon/config/apps/debug.zip", self.download_folder+"/debug.zip")
    print(dt.now().strftime("%Y-%m-%d %H:%M:%S") + ": " + "Debug Files Zipped")`

Mám někde něco špatně? Poradí někdo jak script rozchodit?

ondrejvysek, 2024-06-13:
Opravovat cesty ve skriptu neni uplne dobry napad.
A) AppDaemon je kontejner sam o sobe
B) pokud by nefungovaly cesty, pak by v tom adresari pnd nebylo nic ulozeneho a stazeneho

Co je prosim v tech dvou CSV souborech?

K4BlOs, 2024-06-13:
Děkuji za rychlou odpověď. Zkusil jsem to i bez úprav těch cest ale výsledek je stejný. Stáhne se jen něco. Přikládám csv soubory
[daily-consumption.csv](https://github.com/user-attachments/files/15821115/daily-consumption.csv)
[daily-production.csv](https://github.com/user-attachments/files/15821117/daily-production.csv)

ondrejvysek, 2024-06-13:
Kouknu pozdeji na pocitaci, na prvni pohled je obsah csv obrazek
[image]

ondrejvysek, 2024-06-13:
oba soubory obsahují obrázky - prinscreeny, nevím zdali unikátní, nebo nějaký z těch, co jsou tam vytvořené. V logice problém nevidím, zkusím si nahohodit HA v dockeru, pouze prosím o nějaký bližší popis vaší konfigurace a typu procesoru, abych tomu byl co nejblíže.

K4BlOs, 2024-06-14:
Děkuji za vaší ochotu.
Systém mi běží na HPE ProLiant MicroServer Gen10 Plus G5420 s procesorem Intel(R) Pentium(R) Gold G5420 CPU @ 3.80GHz a 32GB DDR4. Mám tam Debian 11 (aktuálně s verzí jádra Linux 6.1.0-0.deb11.21-amd64).

V Portaineru mám tedy container Home assistenta s namapovaným cestami 
kontainer: /config
host: /home/karel/Docker/homeassistant/config

Dle vašeho návodu mi do HA nešel instalovat doplněk AppDaemon (nevím jestli to neumožňuje verze HA v dockeru nebo to že není supervised). Nainstaloval jsem tedy kontainer AppDaemon, zprovoznil a propojil s HA.
mapování cest v AppDaemonu:
kontainer: /conf
host: /home/karel/Docker/appdaemon/config

Vytvoření tokenu v HA a zadání všech URL cest do AppDaemon, vše proběhlo v pořádku. Spuštění scriptu z HA pomocí run_pnd také. 

Problém je tedy jen to, že script nedoběhne dokonce a skončí chybou viz log výše.

ondrejvysek, 2024-06-14:
Jak jsem psal, appdaemon je sam o sobe kontejner, tj v ha v kontejneru nepobezi a netestoval jsem. Nahodim si nekde amd64 a zkusim. 
To proc to nedobehne je obsah csv souboru, kde je to obrazek a ne data. Ale neni duvod, proc by tam mel byt obrazek obrazovky a ne vlastni data.
Mohu tam dat osetreni chyby, ale to nevyresi absenci vlastnich dat.

ondrejvysek, 2024-06-15:
Tak na to od rána koukám, problém se mi asi podařilo izolovat:
1. soubory se nestahují, pouze se přejmenuje předchozí soubor
2. nejspíše souvisí s nastavením chromium / chromium-driver
3. verze kterou kontejner na deb_amd64 stahuje je 112.0.5615.165
4. vrze která se běžně používá např na tom proxmoxu / raspi,... je 124.0+
5. z nějakého, mě neznámého důvodu, není možné v této verzi nastavit default_download proměnnou a soubory končí kdoví kde
6. nepodařilo se mi připojit na konzoli appdaemon, abych zkusil pohledat tam

Vlastně nyní nevím jak dál a moje znalosti *nix/docker končí :(

ondrejvysek, 2024-06-18:
Koukám na tento problém, můžete prosím použít aktuální verzi skriptu (0.9.9+) je tam debug logování. Vypíše seznam modulů a verzí, prosím o tuto část logu až po chromedriver (náhled a začátek logu na obrázku)

[image]

ondrejvysek, 2024-06-18:
A ještě, zkuste prosím dev appdaemonu `acockburn/appdaemon:dev`. latest je víc jak rok stará, koukal jsem, že by mohlo pomoci (zatím plně netestováno).
akorát je nutné v requirements specifikovat navíc starší numpy:
```
selenium==4.21.0
numpy=1.26.4
pandas==2.2.2
beautifulsoup4==4.12.3
```

K4BlOs, 2024-06-19:
> Koukám na tento problém, můžete prosím použít aktuální verzi skriptu (0.9.9+) je tam debug logování. Vypíše seznam modulů a verzí, prosím o tuto část logu až po chromedriver (náhled a začátek logu [...]

Zde je tedy výpis logu s aktuální verzi scriptu ver = "0.9.9"

2024-06-19 07:28:18: >>>>>>>>>>>> PND Initialize
System Information:
Platform: Linux
Platform Release: 6.1.0-0.deb11.21-amd64
Platform Version: #1 SMP PREEMPT_DYNAMIC Debian 6.1.90-1~bpo11+1 (2024-05-06)
Architecture: x86_64
Processor: 
Python Version: 3.10.11
Installed Python Modules:
Package            Version

aiohttp            3.8.4
aiohttp-jinja2     1.5.1
aiosignal          1.3.1
appdaemon          4.4.2
astral             3.2
async-timeout      4.0.2
attrs              23.2.0
bcrypt             4.0.1
beautifulsoup4     4.12.3
bidict             0.22.1
bs4                0.0.2
certifi            2022.12.7
charset-normalizer 3.1.0
deepdiff           6.3.0
exceptiongroup     1.2.1
feedparser         6.0.10
frozenlist         1.3.3
h11                0.14.0
idna               3.4
iso8601            1.1.0
Jinja2             3.1.2
MarkupSafe         2.1.2
multidict          6.0.4
numpy              1.26.4
ordered-set        4.1.0
outcome            1.3.0.post0
paho-mqtt          1.6.1
pandas             2.2.2
pid                3.0.4
pip                23.0.1
PySocks            1.7.1
python-dateutil    2.8.2
python-engineio    4.4.0
python-socketio    5.8.0
pytz               2023.3
PyYAML             6.0
requests           2.28.2
selenium           4.21.0
setuptools         65.5.1
sgmllib3k          1.0.0
six                1.16.0
sniffio            1.3.1
sockjs             0.11.0
sortedcontainers   2.4.0
soupsieve          2.5
tomli              2.0.1
tomli_w            1.0.0
trio               0.25.1
trio-websocket     0.11.1
typing_extensions  4.12.2
tzdata             2024.1
urllib3            1.26.15
uvloop             0.17.0
websocket-client   1.5.1
wheel              0.40.0
wsproto            1.2.0
yarl               1.8.2
ChromeDriver Version: ChromeDriver 112.0.5615.165 (c262f36e6b1d711ee42d4fbe1343b49960593f18-refs/branch-heads/5615@{#1297})
2024-06-19 07:28:18.581272 INFO AppDaemon: App initialization complete

Zkusím tedy ještě dnes `acockburn/appdaemon:dev` a dám vědět

ondrejvysek, 2024-06-19:
Diky, presne s temito verzemi to vypada na Debianu, na jinych platformach je vyrazne novejsi.

V noci jsem se uz k pretestovani :dev nedostal, ale verze vseho jsou tam novejsi vc. Chrome-driver ktery resi stahovani

K4BlOs, 2024-06-19:
> A ještě, zkuste prosím dev appdaemonu `acockburn/appdaemon:dev`. latest je víc jak rok stará, koukal jsem, že by mohlo pomoci (zatím plně netestováno). akorát je nutné v requirements specifikovat navíc starší numpy:
> 
> ```
> selenium==4.21.0
> numpy==1.26.4
> pandas==2.2.2
> beautifulsoup4==4.12.3
> ```

WAU jste borec!!! nyní tedy již vše funguje! VELKÉ DÍKY !!!
souhrn mého nastavení:

instalace kontaineru `acockburn/appdaemon:dev` s verzemi systémových a python doplňků
[image]

[image]

a následná konfugurace souborů
[image]

[image]

Dále použit `pnd.py` script verze `ver = "0.9.9"` zde jsem zachoval cesty tak jak je máte ve scriptu, ale vyhodilo mi to chybu o nedostupnosti složky
[image]

jelikož mám mapování cest v AppDaemonu takto
kontainer: /conf
host: /home/karel/Docker/appdaemon/config

tak jsem upravil i cesty ve vašem scriptu
[image]

následně tedy spustil v HA run_pnd a dle logu vše proběhlo až do konce

2024-06-19 05:58:58: ********************* Starting 0.9.9 *********************
2024-06-19 05:58:58: ----------------------------------------------
2024-06-19 05:58:58: Hello from AppDaemon for Portal Namerenych Dat
2024-06-19 05:58:59: Driver Loaded
2024-06-19 05:59:00: Website Opened
2024-06-19 05:59:09: App Version: 2.3.5
2024-06-19 05:59:09: H1 tag with text 'Naměřená data' is present.
2024-06-19 05:59:15: Rychla Sestava selected successfully!
2024-06-19 05:59:16: Selecting ELM '792648'
2024-06-19 05:59:16: Valid ELM numbers 'ELM 792648, ELM 792648, ELM 792648, ELM 792648'
2024-06-19 05:59:18: ELM Status:  - 792648
2024-06-19 05:59:18: Iteration 0: Vyhledat Button IS disabled
2024-06-19 05:59:20: ELM Status: ELM 792648 - 792648
2024-06-19 05:59:20: Iteration 1: Vyhledat Button NOT disabled
2024-06-19 05:59:20: Device ELM '792648' selected successfully!
2024-06-19 05:59:21: Button 'Vyhledat data' clicked successfully!
2024-06-19 05:59:23: 07 Profil spotřeby za den (+A)
2024-06-19 05:59:29: Downloading CSV file for 07 Profil spotřeby za den (+A)
2024-06-19 05:59:30: File downloaded and saved as: /conf/apps/pnd/daily-consumption.csv
2024-06-19 05:59:31: 08 Profil výroby za den (-A)
2024-06-19 05:59:36: Downloading CSV file for 08 Profil výroby za den (-A)
2024-06-19 05:59:38: File downloaded and saved as: /conf/apps/pnd/daily-production.csv
2024-06-19 05:59:38: All Done - DAILY DATA DOWNLOADED
2024-06-19 05:59:38: Latest entry: 19.06.2024 23:59 - 0.7445 kWh
2024-06-19 05:59:38: Latest entry: 19.06.2024 23:59 - 36.2507 kWh
2024-06-19 05:59:38: All Done - DAILY DATA PROCESSED
2024-06-19 05:59:39: Data Interval Entered - '27.10.2023 00:00 - 27.10.2024 00:00'
2024-06-19 05:59:41: Selecting 07 Profil spotřeby za den (+A)
2024-06-19 05:59:42: 07 Profil spotřeby za den (+A)
2024-06-19 05:59:45: Exporting data
2024-06-19 05:59:47: File downloaded and saved as: /conf/apps/pnd/range-consumption.csv 14.78 KB
2024-06-19 05:59:47: Selecting 08 Profil výroby za den (-A)
2024-06-19 05:59:47: 08 Profil výroby za den (-A)
2024-06-19 05:59:51: Exporting data
2024-06-19 05:59:52: File downloaded and saved as: /conf/apps/pnd/range-production.csv 14.64 KB
2024-06-19 05:59:52: All Done - INTERVAL DATA DOWNLOADED
2024-06-19 05:59:52: All Done - INTERVAL DATA PROCESSED
2024-06-19 05:59:52: All Done - BROWSER CLOSED
2024-06-19 05:59:52: Sensor State Set to OFF
2024-06-19 05:59:53: Debug Files Zipped
2024-06-19 07:59:53.026140 INFO AppDaemon: pnd: Entity sensor.pnd_script_duration created in namespace: default
2024-06-19 05:59:53: ********************* Duration: 0:00:54.254123 *********************
2024-06-19 05:59:53: ********************* Finished 0.9.9 *********************
2024-06-19 07:59:53.031213 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 55.030995 seconds (limit=10)

ve složce pnd jsou staženy tyto souory
[image]

příklad .csv souboru
[image]

a entity v HA
[image]

Nyní už jen vytvořit dashboard dle vašeho návodu :-)

ondrejvysek, 2024-06-19:
Super! 
Necham zatim otevrene, ty cesty beru jako vylepseni do konfigu, aby se snaze aktualizoval skript v budoucnu.

Pouziju poznatky na rozsireni navodu pro ty, co maji HA v dockeru.

Mozna jestli sem hodite jeste obsah souboru s moduly pro appdaemon, abych nemusel hledat.

K4BlOs, 2024-06-19:
Jestli mislíte obsah souborů `system_packages.txt` a `requirements.txt` tak jsou v postu výše. Ale klidne přihodím znovu
 `system_packages.txt`:
chromium-chromedriver
chromium

a v `requirements.txt`:
selenium==4.21.0
numpy==1.26.4
pandas==2.2.2
beautifulsoup4==4.12.3
bs4

(omlouvám se, nevám jak se sem vkládá kód. Vždy mi to vloží do jedné řádky)

ondrejvysek, 2024-06-19:
Super, takto staci. Dekuju

ondrejvysek, 2024-07-26:
vyřešeno v 0.9.9.1

closed 2024-07-26 by ondrejvysek

waclaw66, 2024-09-16:
Prave jsem narazil na stejny problem a mohu potvrdit, ze pouziti `acockburn/appdaemon:dev` misto `latest` pomohlo.

HellixFix, 2025-01-20:
Nebyl by podrobnější postup pro amatéry?.... dát to dohromady v dockeru na NASu je momentálně trochu problém :-)
díky