ondrejvysek/HomeAssistant-CEZDistribuce-PND #40: Historie

pulss, 2024-06-15:
Dobrý den,

pisete, cituji

Využívá senzory sensor.pnd_consumption a sensor.pnd_production které obsahují denní spotřebu resp výrobu za předchozí den. Senzory jsou třídy (device_class) energy a jsou tedy automaticky ukládány do dlouhodobých dat v HomeAssistant

avsak vidim pouze data za poslednich 8 dni, jak mam nastavenu ukladani historie do databaze v configuration.yaml, delam neco spatne, nebo si mam vytvorit nove senzory pres utility meter abych mel dostupnou celou historii napriklad po mesicich?

Dekuji Pavel

ondrejvysek, 2024-06-15:
Koukám na to, měl jsem za to (dle dokumentace), že energy data jsou z retence vyjmuta. Ona jsou vyjmuta ale pouze ta co jsou v energy dashboardu. Jdu se podívat na možnosti.

pulss, 2024-06-25:
Dobrý den,
je najeky posun, pripadne napad? 
Stacilo by mi udelat si to pres utility meter, ale nevim jak to zapsat cronem, aby stazene data v 2 rano (distribuce mi to zapocitava vzdy v 1:30 a 13:30) se mi zapsaly do predchoziho dne treba s casem 23:59
Dekuji