ondrejvysek/HomeAssistant-CEZDistribuce-PND #41: Unexpected error in worker for App pnd

MarCa123456, 2024-06-17:
Po instalaci se mi nevytvořily tyto tři senzory
[image]

Při běhu PND.py je v logu tato chyba
[image]

ondrejvysek, 2024-06-17:
Co mate prosim za verzi skriptu? (Prvni radek). Toto by melo byt v aktualni verzi osetrene.

MarCa123456, 2024-06-17:
ver = "0.9.7"

ondrejvysek, 2024-06-17:
Je tam aktualizace, ktera by to mela vyresit. Prosim o test a info, zdali zabralo. Diky

MarCa123456, 2024-06-18:
Dobry den,

V mezicase jsem skoncil na problemu popsanem v tomto vlakne

Chyba v Numpy
 #42

Takze pockam az se vyresi toto a dam vedet.

ondrejvysek, 2024-06-18:
Reseni je v tom vlakne, je nutne pridat rucne starsi verzi numpy.

Aktualizace dokumentace dnes odpoledne

MarCa123456, 2024-06-18:
Děkuji. Mohu potvrdit, že problém je nyní vyřešen. Data stažena. Všechny senzory vytvořené tak jak mají. Jen mám teď nějaká nesmyslná data v grafech, jak jsem pustil toho AppDeamona vícekrat za den.
Jdu se podívat jak to promazat. Pokud máte nějaký nápad budu moc rád.

ondrejvysek, 2024-06-18:
V tuto chvíli počkat, za den zmizí

closed 2024-06-18 by ondrejvysek