ondrejvysek/HomeAssistant-CEZDistribuce-PND #42: Chyba v Numpy

jankorous, 2024-06-17:
Pro prvotní instalaci a nastavení podle Readme.md mi zůstává následující chyba:

```
2024-06-17 10:44:19.882820 WARNING AppDaemon: error loading plugin: HASS - ignoring
2024-06-17 10:44:19.883329 WARNING AppDaemon: ------------------------------------------------------------
2024-06-17 10:44:19.891036 WARNING AppDaemon: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/plugin_management.py", line 140, in __init__
    mod = __import__(full_module_name, globals(), locals(), [module_name], 0)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/plugins/hass/hassplugin.py", line 14, in <module>
    from deepdiff import DeepDiff
  File "/usr/lib/python3.11/site-packages/deepdiff/__init__.py", line 10, in <module>
    from .diff import DeepDiff
  File "/usr/lib/python3.11/site-packages/deepdiff/diff.py", line 18, in <module>
    from deepdiff.helper import (strings, bytes_type, numbers, uuids, times, ListItemRemovedOrAdded, notpresent,
  File "/usr/lib/python3.11/site-packages/deepdiff/helper.py", line 63, in <module>
    np_float_ = np.float_
                ^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/numpy/__init__.py", line 397, in __getattr__
    raise AttributeError(
AttributeError: `np.float_` was removed in the NumPy 2.0 release. Use `np.float64` instead.

2024-06-17 10:44:19.891424 WARNING AppDaemon: ------------------------------------------------------------
```

AppDeamon mám podle vzoru.

ondrejvysek, 2024-06-17:
na jaké platformě to běží?
jak vypadá appdaemon.yaml?
jak vypadá konfigurace modulů AppDaemon?
Jaká je verze HA a verze AppDaemon?

Tohle je chyba někde mezi appdaemon a HA.

jankorous, 2024-06-17:
AppDaemon.yaml:

```
---
appdaemon:
  latitude: 52.379189
  longitude: 4.899431
  elevation: 2
  time_zone: Europe/Amsterdam
  thread_duration_warning_threshold: 60
  plugins:
    HASS:
      type: hass
      ha_url: http://192.168.1.242:8123
      token: tádýdádýdá
  app_dir: /homeassistant/appdaemon/apps
http:
  url: http://127.0.0.1:5050
admin:
api:
hadashboard:
logs:
  pnd:
    name: pnd
    filename: /homeassistant/appdaemon/pnd.log
```

apps.yaml:
```
secrets: /homeassistant/secrets.yaml
pnd:
  module: pnd
  class: pnd
  log: pnd
  PNDUserName: !secret pnd_login
  PNDUserPassword: !secret pnd_password
  DataInterval: "01.06.2023 00:00 - 31.06.2024 00:00"
  ELM: "890971"
  DownloadFolder: "/homeassistant/appdaemon/apps/pnd"
```

Core je 2024.6.3 a běží na Rpi 4B 8GB.

Nastavení je takto:
[image]

ondrejvysek, 2024-06-17:
Vypadá normálně. Můžete zkusit restart celého RPi? Občas tam vytuhne nějaká aktualizace. Ale tohle evidentně nestartuje kontejner s AppDaemonem :(

jankorous, 2024-06-17:
I po restartu chyba zůstává a data netečou... 🫣 jako je to jen warning, ale nevypadá, že by se spouštěl.

ondrejvysek, 2024-06-17:
Snip logu by se nenasel? Skript se nepousti sam od sebe, je nutna ta automatizace

jankorous, 2024-06-17:
No to mám jako automatizaci, viz návod:
```
alias: Run PND
description: ""
trigger:
  - platform: time
    at: "00:30:00"
condition: []
action:
  - event: run_pnd
    event_data: {}
mode: single
```

jankorous, 2024-06-17:
Tady ještě konec logu

[a0d7b954_appdaemon_2024-06-17T18-21-37.011Z.log](https://github.com/user-attachments/files/15875440/a0d7b954_appdaemon_2024-06-17T18-21-37.011Z.log)

ondrejvysek, 2024-06-17:
Chyba je porad stejna, cetl jsem vic, napadlo me - zkusit dat prihlasovaci udaje primo do apps.yaml a vyhodit ten secret radek.

Ono to vypada, ze to ma za dalsi aplikaci a muze to na tom umrit. 

Pokus

jankorous, 2024-06-17:
To už jsem tak zkoušel - dat login a pass na přímo, stejná issue 🫣

jankorous, 2024-06-17:
Jj, restartuju pokaždé

jankorous, 2024-06-17:
V GUI toho AppDaemona mám něco vidět?

ondrejvysek, 2024-06-17:
na portu :5050 jeportál appdaemonu, ale tam asi nic nebude, když nestartuje. lze skusit komplet vyčistit apps.yaml, aby byl prázndý? aby zkusil nastartovat bez ničeho
[image]

ondrejvysek, 2024-06-17:
Problém izolován (a projevuje se i v jiných instalacích), jenom nemá rychlé řešení. Někdy v uplynulých dnech došlo k aktualizaci buď Numpy nebo DeeDiff, které jsou podkladovým modulem pro Pandas (čte CSV). Díky této chybě nenaběhne AppDaemon správně a skript nemůže fungovat. Chyba je komplet mimo moji kontrolu.

Zkusím přepsat část s Pandas na jinou komponentu, jenom to nebude dnes, zkusím co nejdřív.

ondrejvysek, 2024-06-17:
Pouze pro referenci a ladeni problemu https://github.com/pandas-dev/pandas/issues/59023

ondrejvysek, 2024-06-17:
Včera (16.6.24) byla vydána verze Numpy 2.0.0 která zásadně mění funkcionalitu, Pandas si s ní zatím nepovída. Je nutné, nežli bude Pandas aktualizovaný používat Numpy 1.26.4. V nastavení AppDaemon je nutné přidat Python `package numpy==1.26.4`
[image]

jankorous, 2024-06-18:
Nice solution, PND se dostalo znatelně dál, nicméně se akce nedaří stáhnout nějaké csv:

```
link 07 Profil spotřeby za den (+A)
link 08 Profil výroby za den (-A)
```

a přijde mi, že po nestažení csv už nezpracuje data nebo se zastaví? Přikládám log.

[a0d7b954_appdaemon_2024-06-18T05-12-36.356Z.log](https://github.com/user-attachments/files/15880926/a0d7b954_appdaemon_2024-06-18T05-12-36.356Z.log)

ondrejvysek, 2024-06-18:
To je asi stejna chyba jako zde https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/issues/39 prosim stahnete ty csv soubory a prejmenujte na png, pokud je v tom obrazek, je to stejny problem.

Jak se pise v tom vlakne, zatim nemam reseni.

Na tom RPi to je managed nebo kontejner?

jankorous, 2024-06-18:
Jj, po přejmenování je to obrázek.

Na RPi mi HA běží jako OS, nic jiného na RPi není.

ondrejvysek, 2024-06-18:
Toto zaviram, udelal jsem novy bug https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/issues/43 at se neplete

closed 2024-06-18 by ondrejvysek