ondrejvysek/HomeAssistant-CEZDistribuce-PND #44: Po aktualizácii HA (Operating System 12.4) sensory - sensory - not found

matomatusov, 2024-06-19:
Po aktualizácii HA (Operating System 12.4) mi prestali fungovať dané senzory.
Nevieš v čom može byť problém?

Dik moc

[image]

--------------------------------------------------------------------------------------
>>>Debug ELM<<<
<div role="group" class="form-group" id="__BVID__66"><label for="window-88185-deviceSet" class="d-block" id="__BVID__66__BV_label_">Množina zařízení</label><div class="bv-no-focus-ring"><div class="pnd-form-select"><div tabindex="-1" class="multiselect"><div class="multiselect__select"></div>  <div class="multiselect__tags"><div class="multiselect__tags-wrap" style="display: none;"></div>  <div class="multiselect__spinner" style="display: none;"></div> <input name="" id="window-88185-deviceSet" type="text" autocomplete="nope" placeholder="Vyberte množinu zařízení" tabindex="0" class="multiselect__input" style="width: 0px; position: absolute; padding: 0px;"> <span class="multiselect__single">ELM 90024314</span> </div> <div tabindex="-1" class="multiselect__content-wrapper" style="max-height: 300px; display: none;"><ul class="multiselect__content" style="display: inline-block;">  <li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option multiselect__option--highlight">Všechny EANy</span> </li><li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option">ELM 10141798</span> </li><li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option multiselect__option--selected">ELM 90024314</span> </li> <li style="display: none;"><span class="multiselect__option">Hledání nevyhovují žádné položky</span></li> <li style="display: none;"><span class="multiselect__option">Žádné položky</span></li> </ul></div></div></div></div></div>
>>>Iteration 0<<<
ELM Span content: 
<div role="group" class="form-group" id="__BVID__66"><label for="window-88185-deviceSet" class="d-block" id="__BVID__66__BV_label_">Množina zařízení</label><div class="bv-no-focus-ring"><div class="pnd-form-select invalid"><div tabindex="-1" class="multiselect"><div class="multiselect__select"></div>  <div class="multiselect__tags"><div class="multiselect__tags-wrap" style="display: none;"></div>  <div class="multiselect__spinner" style="display: none;"></div> <input name="" id="window-88185-deviceSet" type="text" autocomplete="nope" placeholder="Vyberte množinu zařízení" tabindex="0" class="multiselect__input" style="width: 0px; position: absolute; padding: 0px;">  <span class="multiselect__placeholder">
          Vyberte množinu zařízení
        </span></div> <div tabindex="-1" class="multiselect__content-wrapper" style="max-height: 300px; display: none;"><ul class="multiselect__content" style="display: inline-block;">  <li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option">Všechny EANy</span> </li><li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option">ELM 10141798</span> </li><li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option multiselect__option--highlight">ELM 90024314</span> </li> <li style="display: none;"><span class="multiselect__option">Hledání nevyhovují žádné položky</span></li> <li style="display: none;"><span class="multiselect__option">Žádné položky</span></li> </ul></div></div></div></div></div>
>>>Iteration 1<<<
ELM Span content: ELM 90024314
<div role="group" class="form-group" id="__BVID__66"><label for="window-88185-deviceSet" class="d-block" id="__BVID__66__BV_label_">Množina zařízení</label><div class="bv-no-focus-ring"><div class="pnd-form-select"><div tabindex="-1" class="multiselect"><div class="multiselect__select"></div>  <div class="multiselect__tags"><div class="multiselect__tags-wrap" style="display: none;"></div>  <div class="multiselect__spinner" style="display: none;"></div> <input name="" id="window-88185-deviceSet" type="text" autocomplete="nope" placeholder="Vyberte množinu zařízení" tabindex="0" class="multiselect__input" style="width: 0px; position: absolute; padding: 0px;"> <span class="multiselect__single">ELM 90024314</span> </div> <div tabindex="-1" class="multiselect__content-wrapper" style="max-height: 300px; display: none;"><ul class="multiselect__content" style="display: inline-block;">  <li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option">Všechny EANy</span> </li><li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option">ELM 10141798</span> </li><li class="multiselect__element"><span data-select="" data-selected="" data-deselect="" class="multiselect__option multiselect__option--highlight multiselect__option--selected">ELM 90024314</span> </li> <li style="display: none;"><span class="multiselect__option">Hledání nevyhovují žádné položky</span></li> <li style="display: none;"><span class="multiselect__option">Žádné položky</span></li> </ul></div></div></div></div></div>

matomatusov, 2024-06-19:
Collecting typing_extensions>=4.9.0 (from selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/typing_extensions-4.12.2-py3-none-any.whl (37 kB)
Requirement already satisfied: attrs>=23.2.0 in /usr/lib/python3.11/site-packages (from trio~=0.17->selenium) (23.2.0)
Collecting sortedcontainers (from trio~=0.17->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/sortedcontainers-2.4.0-py2.py3-none-any.whl.metadata (10 kB)
Requirement already satisfied: idna in /usr/lib/python3.11/site-packages (from trio~=0.17->selenium) (3.7)
Collecting outcome (from trio~=0.17->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/outcome-1.3.0.post0-py2.py3-none-any.whl (10 kB)
Collecting sniffio>=1.3.0 (from trio~=0.17->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/sniffio-1.3.1-py3-none-any.whl (10 kB)
Requirement already satisfied: wsproto>=0.14 in /usr/lib/python3.11/site-packages (from trio-websocket~=0.9->selenium) (1.2.0)
Collecting PySocks!=1.5.7,<2.0,>=1.5.6 (from urllib3[socks]<3,>=1.26->selenium)
  Downloading https://wheels.home-assistant.io/musllinux-index/PySocks-1.7.1-py3-none-any.whl (16 kB)
Requirement already satisfied: h11<1,>=0.9.0 in /usr/lib/python3.11/site-packages (from wsproto>=0.14->trio-websocket~=0.9->selenium) (0.14.0)
Downloading https://wheels.home-assistant.io/musllinux-index/sortedcontainers-2.4.0-py2.py3-none-any.whl (29 kB)
Installing collected packages: sortedcontainers, typing_extensions, sniffio, PySocks, outcome, trio, trio-websocket, selenium
Successfully installed PySocks-1.7.1 outcome-1.3.0.post0 selenium-4.21.0 sniffio-1.3.1 sortedcontainers-2.4.0 trio-0.25.1 trio-websocket-0.11.1 typing_extensions-4.12.2
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting pandas
  Downloading pandas-2.2.2-cp311-cp311-musllinux_1_1_x86_64.whl.metadata (19 kB)
Collecting numpy>=1.23.2 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/numpy-2.0.0-cp311-cp311-musllinux_1_2_x86_64.whl (11.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.5/11.5 MB 6.2 MB/s eta 0:00:00
Requirement already satisfied: python-dateutil>=2.8.2 in /usr/lib/python3.11/site-packages (from pandas) (2.8.2)
Requirement already satisfied: pytz>=2020.1 in /usr/lib/python3.11/site-packages (from pandas) (2023.3.post1)
Collecting tzdata>=2022.7 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/tzdata-2024.1-py2.py3-none-any.whl (345 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 345.4/345.4 kB 6.5 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3.11/site-packages (from python-dateutil>=2.8.2->pandas) (1.16.0)
Downloading pandas-2.2.2-cp311-cp311-musllinux_1_1_x86_64.whl (13.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.9/13.9 MB 5.5 MB/s eta 0:00:00
Installing collected packages: tzdata, numpy, pandas
Successfully installed numpy-2.0.0 pandas-2.2.2 tzdata-2024.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting bs4
  Downloading https://wheels.home-assistant.io/musllinux-index/bs4-0.0.2-py2.py3-none-any.whl.metadata (411 bytes)
Collecting beautifulsoup4 (from bs4)
  Downloading https://wheels.home-assistant.io/musllinux-index/beautifulsoup4-4.12.3-py3-none-any.whl (147 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 147.9/147.9 kB 2.6 MB/s eta 0:00:00
Collecting soupsieve>1.2 (from beautifulsoup4->bs4)
  Downloading https://wheels.home-assistant.io/musllinux-index/soupsieve-2.5-py3-none-any.whl (36 kB)
Downloading https://wheels.home-assistant.io/musllinux-index/bs4-0.0.2-py2.py3-none-any.whl (1.2 kB)
Installing collected packages: soupsieve, beautifulsoup4, bs4
Successfully installed beautifulsoup4-4.12.3 bs4-0.0.2 soupsieve-2.5
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[20:36:36] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-06-18 20:36:40.498544 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-06-18 20:36:40.498824 INFO AppDaemon: Python version is 3.11.9
2024-06-18 20:36:40.499045 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-06-18 20:36:40.499203 INFO AppDaemon: Added log: AppDaemon
2024-06-18 20:36:40.499363 INFO AppDaemon: Added log: Error
2024-06-18 20:36:40.499508 INFO AppDaemon: Added log: Access
2024-06-18 20:36:40.499661 INFO AppDaemon: Added log: Diag
2024-06-18 20:36:40.499855 INFO AppDaemon: Added log: pnd
2024-06-18 20:36:40.740460 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
[... cut ...]
2024-06-18 20:36:41.948974 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2024-06-18 20:36:41.949556 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pnd to module import path
2024-06-18 20:36:41.953868 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pnd.py
2024-06-18 20:36:42.918837 INFO AppDaemon: Loading app pnd using class pnd from module pnd
2024-06-18 20:36:42.920224 INFO AppDaemon: Calling initialize() for pnd
2024-06-18 20:36:42: >>>>>>>>>>>> PND Initialize
2024-06-18 20:36:42.923638 INFO AppDaemon: App initialization complete

ondrejvysek, 2024-06-19:
Prosím viz zde https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/issues/42#issuecomment-2174487719 Dnes aktualizuji dokumentaci

matomatusov, 2024-06-19:
Ahoj, dik za info. Funguje. V grafe PND Včerajší stav mám null.

ondrejvysek, 2024-06-19:
Super! Bohužel zpětně nelze. Nicméně aktuálně senzory jsou uchovány dle nastavení retence (obvykle 10 dní), řeším jak je nechat na dobu neurčitou.