ondrejvysek/HomeAssistant-CEZDistribuce-PND #45: WARNING pnd: ------------------------------------------------------------

kyselyna, 2024-07-04:
Prosím o pomoct po aktualizaci přestalo fungovat nevíte čím to může být?? LOG níže.  Děkuji

HA:
Core   2024.7.0
Supervisor   2024.06.2
Operating System   12.4
Frontend   20240703.0

LOG:
2024-07-04 11:02:51: ********************* Starting 0.9.6 *********************
2024-07-04 11:02:51: ----------------------------------------------
2024-07-04 11:02:51: Hello from AppDaemon for Portal Namerenych Dat
2024-07-04 11:02:52: Driver Loaded
2024-07-04 11:02:54: Website Opened
2024-07-04 11:03:09: App Version: 2.4.1
2024-07-04 11:03:16: H1 tag with text 'Naměřená data' is present.
2024-07-04 11:03:17.398061 WARNING pnd: ------------------------------------------------------------
2024-07-04 11:03:17.400350 WARNING pnd: Unexpected error in worker for App pnd:
2024-07-04 11:03:17.401382 WARNING pnd: Worker Ags: {'id': '4f78df2c0ade4af1af86dedd4419f7e9', 'name': 'pnd', 'objectid': 'c5e7ae0241654011a332c28e8d87db22', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7fa3f28b50>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2024-07-04T09:02:51.747138+00:00', 'context': {'id': '01J1YFPWB06DV22NFEZ5WPKHHC', 'parent_id': '01J1YFPWAZ39YNVSJNN0DQZBA8', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2024-07-04 11:03:17.401944 WARNING pnd: ------------------------------------------------------------
2024-07-04 11:03:17.411084 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/config/apps/pnd.py", line 221, in run_pnd
    tabulka_dat_button.click()
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/webelement.py", line 94, in click
    self._execute(Command.CLICK_ELEMENT)
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/webelement.py", line 395, in _execute
    return self._parent.execute(command, params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/webdriver.py", line 354, in execute
    self.error_handler.check_response(response)
  File "/usr/lib/python3.11/site-packages/selenium/webdriver/remote/errorhandler.py", line 229, in check_response
    raise exception_class(message, screen, stacktrace)
selenium.common.exceptions.ElementClickInterceptedException: Message: element click intercepted: Element <button title="Export" type="button" aria-pressed="true" autocomplete="off" class="btn pnd-btn btn-outline-secondary active">...</button> is not clickable at point (343, 266). Other element would receive the click: <div id="__BVID__34" role="dialog" aria-describedby="__BVID__34___BV_modal_body_" class="modal fade show" aria-modal="true" style="display: block;">...</div>
  (Session info: chrome-headless-shell=124.0.6367.78)

2024-07-04 11:03:17.411517 WARNING pnd: ------------------------------------------------------------

AppDeamon Settings:
[image]

kyselyna, 2024-07-04:
Je možné že to je tímto? 

[image]

kyselyna, 2024-07-04:
Vyřešeno problém byl celou dobu v hlášce co se objevila na Stránce Čez Distribuce po jejím odkliknutí vše zase běží tak jak má.

czgif, 2024-07-04:
Ahoj,

teď jsem taky chtěl napsat ohledně této hlášky. Už se mi to stalo 2x že mi nafungovalo načítání dat a vždy tam bylo info o nedostupnosti portálu,  po přihlášenjeho potvrzení opět načítání dat funkční. Možná by se toto dalo ošetřit ve scriptu :) ???

ondrejvysek, 2024-07-04:
Dalo :) jenom se to musi dit kdyz u toho jsem :) po jednom odkliknuti to zmizi a html kod se neda kde vzit.