ondrejvysek/HomeAssistant-CEZDistribuce-PND #46: Chyba pri inicializaci

tkupka, 2024-07-09:
Pri inicializaci aplikace skoci s chybou inicialize chrome driveru:
````
2024-07-09 09:42:55.232461 INFO AppDaemon: Running on port 5050
2024-07-09 09:42:55.272092 INFO HASS: Evaluating startup conditions
2024-07-09 09:42:55.279735 INFO HASS: Startup condition met: hass state=RUNNING
2024-07-09 09:42:55.280227 INFO HASS: All startup conditions met
2024-07-09 09:42:55.428127 INFO AppDaemon: Got initial state from namespace default
2024-07-09 09:42:57.282572 INFO AppDaemon: Scheduler running in realtime
2024-07-09 09:42:57.290140 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2024-07-09 09:42:57.291104 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pnd to module import path
2024-07-09 09:42:57.299842 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/stateLogger.py - ignoring
2024-07-09 09:42:57.301450 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pv_regulation.py
2024-07-09 09:42:57.379033 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/hello.py
2024-07-09 09:42:57.393900 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/growattServer.py - ignoring
2024-07-09 09:42:57.395339 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/growatt_app.py
2024-07-09 09:42:57.432964 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/PvData.py - ignoring
2024-07-09 09:42:57.434845 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/PvPowerCalculation.py - ignoring
2024-07-09 09:42:57.438197 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pnd/pnd.py
2024-07-09 09:42:57.446224 WARNING Error: ------------------------------------------------------------
2024-07-09 09:42:57.446722 WARNING Error: Unexpected error loading module: /homeassistant/appdaemon/apps/pnd/pnd.py:
2024-07-09 09:42:57.447153 WARNING Error: ------------------------------------------------------------
2024-07-09 09:42:57.453835 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/homeassistant/appdaemon/apps/pnd/pnd.py", line 151
    "status": "ERROR: Nepoda�ilo se inicializovat Chrome Driver, zkontroluj nastaven� AppDaemon",
                                                                                            ^
SyntaxError: (unicode error) 'utf-8' codec can't decode byte 0xf8 in position 13: invalid start byte

2024-07-09 09:42:57.454486 WARNING Error: ------------------------------------------------------------
2024-07-09 09:42:57.454959 WARNING AppDaemon: Removing associated apps:
2024-07-09 09:42:57.455532 WARNING AppDaemon: pnd
```

ondrejvysek, 2024-07-09:
Jake je prosim nastaveni appdaemon? Jaka verze? Kde bezi? (Na HA nebo samostatne v kontejneru)

tkupka, 2024-07-09:
Bezi to na RPI4 

```
Raspberry Pi 4

rpi4-64
```
V kontejneru
```
-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.6
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.7.1
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
```
config:
```
secrets: /homeassistant/secrets.yaml
appdaemon:
  latitude: 52.379189
  longitude: 4.899431
  elevation: 2
  time_zone: Europe/Amsterdam
  app_dir: /homeassistant/appdaemon/apps
  plugins:
    HASS:
      type: hass
      ha_url: https://internal.vlacek.cz:8123
      token: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJkNDMyY2NjYTQ4NjE0MDY1ODgwMDhkOTVjNGI1ZGQyZSIsImlhdCI6MTcwOTQ2MzI0MywiZXhwIjoyMDI0ODIzMjQzfQ.fUaqhhddCrtii6RlnjBm9VLL3xa1x7yzytD3IjRzHwI
      cert_verify: false
http:
  url: http://127.0.0.1:5050
  transport: socketio
admin:
api:
hadashboard:
logs:
  pnd:
    name: pnd
    filename: /homeassistant/appdaemon/pnd.log
```

ondrejvysek, 2024-07-10:
Můžete ještě použít aktuální verzi skriptu, při inicializaci píše moduly a verze (stačí aktualizovat skript, init proběhne bez restartu AppDaemon) - o to bych ještě poprosil

tkupka, 2024-07-11:
Posledni zmeny jsou ve skriptu pred 3mi mesici a tim ze jsem to nainstaloval nedavno tak mam posledni verzi (0.9.9). Jinak v instalaci nemate zadne info ze je potreba nekde registrovat senzory. Divil jsem se jak je vytvarite, kdyz mate jenom python script  ;-). 
Jinak staci `sensors.yml` nakopcit do adresare napr packages, ktery mate includnuty v `configuration.yml` to vetsinou je kdyz ma clovek tech aplikaci vice
```
homeassistant:
  packages: !include_dir_named packages
```
Dalsi problem je ze printf nejde do logu na to je treba `self.log("Initializing...")` napr. Ale melo by to byt konzoli appdaeomnu a tam nic neni .

jelikoz nikde nebyl zadny log tak to bylo divne... A problem je kompilacniho razu jelikoz mate tam ceske znaky a ja ten `pnd.py` nemel v UTF-8, coz jsem si nevsim jeklikoz se mi ceske znaky zobrazovaly normalne.

ondrejvysek, 2024-07-11:
Nevím zdali úplně rozumím. Je problém vyřešený?
Díky za komentáře:
ad senzory: není to nikde zmíněné záměrně, senzor se skriptem vytváří nespravovaný, sensor.yaml,... je experimentální, dle každého potřeby neb každý vytváří senzory jinak a jiným způsobem. Jak budu mít vyřešené dlouhodobé ukládání doplním.
ad log: záměrně byl self.log odebrán a nahrazen, aby se vše psalo do konzole appdaemonu, díky tomu, že je to vlastní kontejner, je lepší orientace v informacích při řešení problémů - což je v dokumentaci napsané
ad UTF-8: i pokud tam nebudou české znaky (a pokud tam jsou odeberu) tak soubor asi UTF-8 zůstane.

Jinak Log AppDaemonu s logem ze skriptu funguje bez problémů:
[image]

tkupka, 2024-07-12:
* UTF-8 to je v pohde to byla moje blbost, ze jsem si nevsim.
* log mate pravdu ze je to  reseni problemu napsane ze to jde do konzole, ale v v nastaveni je v `appdaemon.yaml`  ref na soubor tak jsem predpokladal ze log jde tam.

Nakonec se to nastavit povedlo. 
Ale pri stahovani dat nastala chyba
[image]
Vzy ten posledni csv soubor je stazeny png obrazek

Viz;
[daily-production.csv](https://github.com/user-attachments/files/16199540/daily-production.csv)

Log:
[debug-ELM.txt](https://github.com/user-attachments/files/16199546/debug-ELM.txt)

Error:
```
2024-07-12 23:49:55.712099 WARNING pnd: ------------------------------------------------------------
2024-07-12 23:54:23.393978 WARNING pnd: ------------------------------------------------------------
2024-07-12 23:54:23.395005 WARNING pnd: Unexpected error in worker for App pnd:
2024-07-12 23:54:23.395989 WARNING pnd: Worker Ags: {'id': '7e006caf506444bfb631bbecd466183e', 'name': 'pnd', 'objectid': '2dbe2bf4522f48b2867d8615aae7e2ac', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f81902890>>, 'data': {'metadata': {'origin': 'REMOTE', 'time_fired': '2024-07-12T21:52:49.060451+00:00', 'context': {'id': '01J2MEYEX4DW3T4ND6WQXP1DCJ', 'parent_id': None, 'user_id': 'a3ade110a708453d9ee9b21cb6423c46'}}}, 'pin_app': True, 'pin_thread': 3, 'kwargs': {'__thread_id': 'thread-3'}}
2024-07-12 23:54:23.396933 WARNING pnd: ------------------------------------------------------------
2024-07-12 23:54:23.412584 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/homeassistant/appdaemon/apps/pnd/pnd.py", line 513, in run_pnd
    data_production = pd.read_csv(self.download_folder + '/daily-production.csv', delimiter=';', encoding='latin1')
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/io/parsers/readers.py", line 1026, in read_csv
    return _read(filepath_or_buffer, kwds)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/io/parsers/readers.py", line 626, in _read
    return parser.read(nrows)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/io/parsers/readers.py", line 1923, in read
    ) = self._engine.read(  # type: ignore[attr-defined]
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/io/parsers/c_parser_wrapper.py", line 234, in read
    chunks = self._reader.read_low_memory(nrows)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "parsers.pyx", line 838, in pandas._libs.parsers.TextReader.read_low_memory
  File "parsers.pyx", line 905, in pandas._libs.parsers.TextReader._read_rows
  File "parsers.pyx", line 874, in pandas._libs.parsers.TextReader._tokenize_rows
  File "parsers.pyx", line 891, in pandas._libs.parsers.TextReader._check_tokenize_status
  File "parsers.pyx", line 2061, in pandas._libs.parsers.raise_parser_error
pandas.errors.ParserError: Error tokenizing data. C error: Expected 1 fields in line 4, saw 2

2024-07-12 23:54:23.414453 WARNING pnd: ------------------------------------------------------------

```

tkupka, 2024-07-12:
Jo a jeste mala poznamka. Jelikoz `DownloadFolder: "/homeassistant/appdaemon/apps/pnd"` je spolecny se scriptem, tak po supsteni vsecko smazlo vcetne `pnd.py` ;-)
```
delete_folder_contents(self.download_folder+"/")    
```

tkupka, 2024-07-12:
dalsi run skoncil trochu jinak:
[image]

tkupka, 2024-07-12:
Jakmile portal bezi tak vsecko v poradku dobehne:
```
2024-07-13 00:18:53: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/data/range-production.csv 13.25 KB
2024-07-13 00:18:53: All Done - INTERVAL DATA DOWNLOADED
2024-07-13 00:18:53: All Done - INTERVAL DATA PROCESSED
2024-07-13 00:18:53: All Done - BROWSER CLOSED
2024-07-13 00:18:53: Sensor State Set to OFF
2024-07-13 00:18:54: Debug Files Zipped
2024-07-13 00:18:54: ********************* Duration: 0:01:53.589890 *********************
2024-07-13 00:18:54: ********************* Finished 0.9.9 *********************
```

ondrejvysek, 2024-07-13:
OK, zkusim všechny odpovědi :)

1. data stažená jako obrázek - není portálem, ale za nějakých okolností (nevím jakých - není žádná přímá souvislost s ničím) funkce wait for download vrátí obrázek. mám tu pracovní verzi, hodim dnes / zítra, která nemá tento wait, ale čekání natvrdo a problém se neprojevuje.
2. adresář pro data a skript by neměl být společný. sktipt je /appdaemon/apps data pak /appdaemon/apps/pnd
3. vychátání chybových stavů portálů zkouším postupně, ale ne vždy se podaří chytit ve chvíli, kdy má portál nějakou bolístku, aby se dalo odladit do skriptu.

tkupka, 2024-07-14:
add 1 to se asi udela selenium automaticky pri chybe aby se dalo detekovat v cem je problem. V mojem pripade to bylo tim ze portal je pomaly a nestily se nahrat data
add 2 . to byla moje chyba mam tech aplikaci vice tak jsem si v tom chtel udelat poradek a zkonfiguroval jsem to blbe

Takze v podstate je vsecko OK.

Slo by do skriptu pridat stazeni dat ` 00 Profil spotřeby a výroby + Rv ` bylo by to fajn pro korelaci lokalniho mereni a toho co nameril CEZ

closed 2024-07-26 by ondrejvysek