ondrejvysek/HomeAssistant-CEZDistribuce-PND #48: Error - po updatu Home Assistant Operating System 13.0

matomatusov, 2024-08-14:
Po updatu HA Operating systém na 13.0 nemožen nečítať dáta z PND.

[image]

Log z  APPDaemon
```
2024-08-14 14:38:53: ********************* Starting 0.9.9.1 *********************
2024-08-14 14:38:53.197289 INFO AppDaemon: pnd: Entity binary_sensor.pnd_running created in namespace: default
2024-08-14 14:38:53: ----------------------------------------------
2024-08-14 14:38:53: Hello from AppDaemon for Portal Namerenych Dat
2024-08-14 14:39:00: Driver Loaded
2024-08-14 14:39:03: Website Opened
2024-08-14 14:39:16: H1 tag with text 'Naměřená data' is present.
2024-08-14 14:39:16: Modal dialog not found. Continuing without closing modal.
2024-08-14 14:39:18: App Version: 2.4.1
2024-08-14 14:39:24: Rychla Sestava selected successfully!
2024-08-14 14:39:26: Selecting ELM '90024314'
2024-08-14 14:39:26: Valid ELM numbers 'ELM 10141798, ELM 90024314'
2024-08-14 14:39:29: ELM Status:  - 90024314
2024-08-14 14:39:29: Iteration 0: Vyhledat Button IS disabled
2024-08-14 14:39:31: ELM Status: ELM 90024314 - 90024314
2024-08-14 14:39:31: Iteration 1: Vyhledat Button NOT disabled
2024-08-14 14:39:31: Device ELM '90024314' selected successfully!
2024-08-14 14:39:33: Button 'Vyhledat data' clicked successfully!
2024-08-14 14:39:37: 07 Profil spotřeby za den (+A)
2024-08-14 14:39:44: Downloading CSV file for 07 Profil spotřeby za den (+A)
2024-08-14 14:39:49: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-consumption.csv
2024-08-14 14:39:50: 08 Profil výroby za den (-A)
2024-08-14 14:39:56: Downloading CSV file for 08 Profil výroby za den (-A)
2024-08-14 14:40:02: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-production.csv
2024-08-14 14:40:02: All Done - DAILY DATA DOWNLOADED
2024-08-14 14:40:02: Latest entry: 14.08.2024 23:59 - 1.143 kWh
2024-08-14 14:40:02: Latest entry: 14.08.2024 23:59 - 38.9228 kWh
2024-08-14 14:40:02: All Done - DAILY DATA PROCESSED
2024-08-14 14:40:06: Data Interval Entered - '01.01.2024 00:00 - 31.12.2024 23:59'
2024-08-14 14:40:09: Selecting 07 Profil spotřeby za den (+A)
2024-08-14 14:40:09: 07 Profil spotřeby za den (+A)
2024-08-14 14:40:14: ERROR: Failed to find link 07 Profil spotřeby za den (+A)
2024-08-14 14:40:15: Exporting data
2024-08-14 14:40:18: ERROR: Failed to download CSV file for 07 Profil spotřeby za den (+A)
2024-08-14 14:40:23.880556 WARNING pnd: ------------------------------------------------------------
2024-08-14 14:40:23.881059 WARNING pnd: Unexpected error in worker for App pnd:
2024-08-14 14:40:23.881368 WARNING pnd: Worker Ags: {'id': 'ba150b9f496842009657505b9d495a63', 'name': 'pnd', 'objectid': '0b46ddafa45f4567991bc7db763fc63c', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f566d230f50>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2024-08-14T12:38:53.158108+00:00', 'context': {'id': '01J58EA2C2810601W13JA3WFZN', 'parent_id': '01J58EA2C2DPW7F7RDY6GH40TX', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2024-08-14 14:40:23.882757 WARNING pnd: ------------------------------------------------------------
2024-08-14 14:40:23.904574 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/homeassistant/appdaemon/apps/pnd.py", line 702, in run_pnd
    os.rename(downloaded_file, new_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/homeassistant/appdaemon/apps/pnd/pnd_export.csv' -> '/homeassistant/appdaemon/apps/pnd/range-consumption.csv'

2024-08-14 14:40:23.906369 WARNING pnd: ------------------------------------------------------------
2024-08-14 14:40:24.052283 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 91.04946 seconds (limit=60.0)
```

matomatusov, 2024-08-14:
Podarilo sa mi to rozchodiť.

closed 2024-08-14 by matomatusov

ondrejvysek, 2024-08-14:
Jak se podařilo rozchodit? pokud by se někomu dalšímu vyskytlo, pomohlo by :)

matomatusov, 2024-08-14:
Pomohol mi reštart HA a asi na 3 krát keď som pustil skript sa mi dáta obnovili.