ondrejvysek/HomeAssistant-CEZDistribuce-PND #49: pnd.py

Dawerius, 2024-08-23:
Nějak nedojede skript

 File "/homeassistant/appdaemon/apps/pnd.py", line 128
    <title>HomeAssistant-CEZDistribuce-PND/pnd.py at main · ondrejvysek/HomeAssistant-CEZDistribuce-PND · GitHub</title>
                                                          ^
SyntaxError: invalid character '·' (U+00B7)er

VlcaCZ, 2024-08-23:
Mohu potvrdit. 
V úterý jsem instaloval na čistý HA a nepodařilo se. 
Myslel jsem že jsem někde udělal chybu, ale je chyba jinde.

ondrejvysek, 2024-08-23:
Kouknu pristi tyden, mezi tim prosim verze HA a debug info, ktere skript napise pri svem startu (ruzne verze modulu...)

Dawerius, 2024-08-23:
Home Assistant

Core
2024.8.2
Supervisor
2024.08.0
Operating System
13.1
Frontend
20240809.0

2024-08-23 12:34:17.505674 INFO AppDaemon: App 'pnd' added
2024-08-23 12:34:17.508588 INFO AppDaemon: Found 1 active apps
2024-08-23 12:34:17.510054 INFO AppDaemon: Found 0 inactive apps
2024-08-23 12:34:17.511015 INFO AppDaemon: Found 0 global libraries
2024-08-23 12:34:17.512542 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-08-23 12:34:17.515365 INFO AppDaemon: Running on port 5050
2024-08-23 12:34:17.562873 INFO HASS: Evaluating startup conditions
2024-08-23 12:34:17.571557 INFO HASS: Startup condition met: hass state=RUNNING
2024-08-23 12:34:17.572490 INFO HASS: All startup conditions met
2024-08-23 12:34:17.605971 INFO AppDaemon: Got initial state from namespace default
2024-08-23 12:34:19.571922 INFO AppDaemon: Scheduler running in realtime
2024-08-23 12:34:19.577887 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2024-08-23 12:34:19.579113 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pnd to module import path
2024-08-23 12:34:19.583127 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pnd.py
2024-08-23 12:34:19.595257 WARNING Error: ------------------------------------------------------------
2024-08-23 12:34:19.596147 WARNING Error: Unexpected error loading module: /homeassistant/appdaemon/apps/pnd.py:
2024-08-23 12:34:19.596984 WARNING Error: ------------------------------------------------------------
2024-08-23 12:34:19.606485 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/homeassistant/appdaemon/apps/pnd.py", line 128
    <title>HomeAssistant-CEZDistribuce-PND/pnd.py at main · ondrejvysek/HomeAssistant-CEZDistribuce-PND · GitHub</title>
                                                          ^
SyntaxError: invalid character '·' (U+00B7)

2024-08-23 12:34:19.607450 WARNING Error: ------------------------------------------------------------
2024-08-23 12:34:19.608546 WARNING AppDaemon: Removing associated apps:
2024-08-23 12:34:19.609729 WARNING AppDaemon: pnd
2024-08-23 12:34:19.611609 INFO AppDaemon: App initialization complete

ondrejvysek, 2024-08-23:
Diky, evidentne chyba nekde u inicializace. Jak bude oprava, dam vedet (v druhe pulce pristiho tydne)

ondrejvysek, 2024-08-24:
@Dawerius @VlcaCZ nebyl proveden i upgrade na HACS 2.0?

Dawerius, 2024-08-24:
Ano. HACS 2.0

ondrejvysek, 2024-08-30:
V konfiguraci:
Core 2024.8.3
Supervisor 2024.08.0
Operating System 13.1
Frontend 20240809.0

běží skript ok

Aktualizace HACS 2.0 běží také ok.

jak jste @Dawerius @VlcaCZ kopírovali skript? píše to chybu na řádce 128
```
File "/homeassistant/appdaemon/apps/pnd.py", **line 128**
<title>HomeAssistant-CEZDistribuce-PND/pnd.py at main · ondrejvysek/HomeAssistant-CEZDistribuce-PND · GitHub</title>

```
nicméně řádka 128 obsahuje `os.makedirs(self.download_folder, exist_ok=True)` nikoliv výše uvedený HTML kód. prosím zkuste skript zkopírovat znovu bez [HTML formátování přímo](https://raw.githubusercontent.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/main/pnd.py)

VlcaCZ, 2024-08-30:
> V konfiguraci: Core 2024.8.3 Supervisor 2024.08.0 Operating System 13.1 Frontend 20240809.0
>
> běží skript ok
>
> Aktualizace HACS 2.0 běží také ok.
>
> jak jste @Dawerius @VlcaCZ kopírovali skript? píše to [...]

V neděli budu instalovat nový HW, tak dám vědět. 
Díky moc ;)

ondrejvysek, 2024-09-02:
@VlcaCZ @Dawerius mohu uzavřít nebo chyba stále přetrvává?

VlcaCZ, 2024-09-02:
> @VlcaCZ @Dawerius mohu uzavřít nebo chyba stále přetrvává?

Ještě jsem bohuzel nestihl otestovat. Mám potíže s HW

ondrejvysek, 2024-09-15:
Pro neaktivitu uzavírám

closed 2024-09-15 by ondrejvysek