ondrejvysek/HomeAssistant-CEZDistribuce-PND #50: chyba pri stahovani pnd_export.csv

waclaw66, 2024-09-15:
Ahoj, dela mi to na dvou ruznych HA...
```
2024-09-15 11:57:59.182720 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/homeassistant/appdaemon/apps/pnd.py", line 702, in run_pnd
    os.rename(downloaded_file, new_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/homeassistant/appdaemon/apps/pnd/pnd_export.csv' -> '/homeassistant/appdaemon/apps/pnd/range-consumption.csv'
```

0.9.9.1

ondrejvysek, 2024-09-15:
Ahoj,
1. Prosím o výpis verzí / modulů, co se vypíše po restartu AppDaemon
2. zkusmo odkomentovat řádek 696 a komentovat řádek 697, výsledek bude tedy takto
```
    downloaded_file = wait_for_download(self.download_folder)
    #downloaded_file = os.path.join(self.download_folder, "pnd_export.csv")
```

Problém se občas objeví, při opětovném spuštění skriptu je obvykle ok. Bohužel nezabírají různé délky timeout,... hodně závisí na verzi chromedriver

waclaw66, 2024-09-15:
Po prehozeni zakomentovani se to dostalo trochu dale, ale stejne bez uspechu...
```
2024-09-15 12:40:15: ********************* Starting 0.9.9.1 *********************
2024-09-15 12:40:15: ----------------------------------------------
2024-09-15 12:40:15: Hello from AppDaemon for Portal Namerenych Dat
2024-09-15 12:40:16: Driver Loaded
2024-09-15 12:40:25: Website Opened
2024-09-15 12:40:44: H1 tag with text 'Naměřená data' is present.
2024-09-15 12:40:44: Modal dialog not found. Continuing without closing modal.
2024-09-15 12:40:47: App Version: 2.4.1
2024-09-15 12:40:52: Rychla Sestava selected successfully!
2024-09-15 12:40:54: Selecting ELM '30000xxxxx'
2024-09-15 12:40:55: Valid ELM numbers 'ELM 900xxxxx, ELM 30000xxxxx'
2024-09-15 12:40:58: ELM Status:  - 30000xxxxx
2024-09-15 12:40:58: Iteration 0: Vyhledat Button IS disabled
2024-09-15 12:41:01: ELM Status: ELM 30000xxxxx - 30000xxxxx
2024-09-15 12:41:01: Iteration 1: Vyhledat Button NOT disabled
2024-09-15 12:41:01: Device ELM '30000xxxxx' selected successfully!
2024-09-15 12:41:03: Button 'Vyhledat data' clicked successfully!
2024-09-15 12:41:07: 07 Profil spotřeby za den (+A)
2024-09-15 12:41:15: Downloading CSV file for 07 Profil spotřeby za den (+A)
2024-09-15 12:41:21: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-consumption.csv
2024-09-15 12:41:21: 08 Profil výroby za den (-A)
2024-09-15 12:41:29: Downloading CSV file for 08 Profil výroby za den (-A)
2024-09-15 12:41:34: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-production.csv
2024-09-15 12:41:34: All Done - DAILY DATA DOWNLOADED
2024-09-15 12:41:34: Latest entry: 15.09.2024 23:59 - 8.617 kWh
2024-09-15 12:41:34: Latest entry: 15.09.2024 23:59 - 0.147 kWh
2024-09-15 12:41:34: All Done - DAILY DATA PROCESSED
2024-09-15 12:41:38: Data Interval Entered - '01.01.2024 00:00 - 01.01.2030 00:00'
2024-09-15 12:41:41: Selecting 07 Profil spotřeby za den (+A)
2024-09-15 12:41:41: 07 Profil spotřeby za den (+A)
2024-09-15 12:41:46: Exporting data
2024-09-15 12:41:55: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/range-consumption.csv 137.68 KB
2024-09-15 12:41:55: Selecting 08 Profil výroby za den (-A)
2024-09-15 12:41:55: ERROR: Failed to find link 08 Profil výroby za den (-A)
2024-09-15 12:41:55: Exporting data
2024-09-15 12:42:05: ERROR: Failed to download CSV file for 08 Profil výroby za den (-A)
2024-09-15 12:42:10.262678 WARNING pnd: ------------------------------------------------------------
2024-09-15 12:42:10.263723 WARNING pnd: Unexpected error in worker for App pnd:
2024-09-15 12:42:10.265073 WARNING pnd: Worker Ags: {'id': '82490f5094284a53b8d1b3b8fe67539c', 'name': 'pnd', 'objectid': 'a444e6bda1a34d07bcc646c45cefe66e', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f7651c610>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2024-09-15T10:40:15.141660+00:00', 'context': {'id': '01J7TM9NRXNV29EQVBCY0RMP2Z', 'parent_id': '01J7TM9NRV8VNZ5WWEAJKNM9BW', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2024-09-15 12:42:10.265797 WARNING pnd: ------------------------------------------------------------
2024-09-15 12:42:10.284091 WARNING pnd: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/homeassistant/appdaemon/apps/pnd.py", line 774, in run_pnd
    os.rename(downloaded_file, new_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/homeassistant/appdaemon/apps/pnd/pnd_export.csv' -> '/homeassistant/appdaemon/apps/pnd/range-production.csv'

2024-09-15 12:42:10.284490 WARNING pnd: ------------------------------------------------------------
2024-09-15 12:42:10.350247 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 115.349626 seconds (limit=10)
```

Vypis modulu:
```
2024-09-15 12:33:43: >>>>>>>>>>>> PND Initialize
System Information:
Platform: Linux
Platform Release: 6.6.31-haos-raspi
Platform Version: #1 SMP PREEMPT Wed Aug 21 16:30:30 UTC 2024
Architecture: aarch64
Processor: 
Python Version: 3.11.9

Installed Python Modules:
Package            Version
------------------ ------------
aiohttp            3.8.6
aiohttp-jinja2     1.5.1
aiosignal          1.3.1
[... 36 lines cut ...]
setuptools         68.2.2
sgmllib3k          1.0.0
simple-websocket   1.0.0
six                1.16.0
sniffio            1.3.1
sockjs             0.11.0
sortedcontainers   2.4.0
soupsieve          2.6
tomli              2.0.1
tomli_w            1.0.0
trio               0.26.2
trio-websocket     0.11.1
typing_extensions  4.12.2
tzdata             2024.1
urllib3            1.26.18
uvloop             0.17.0
websocket-client   1.8.0
wheel              0.42.0
wsproto            1.2.0
yarl               1.9.4

ChromeDriver Version: ChromeDriver 124.0.6367.78 (a087f2dd364ddd58b9c016ef1bf563d2bc138711-refs/branch-heads/6367@{#954})
2024-09-15 12:33:50.225030 INFO AppDaemon: App initialization complete

```

ondrejvysek, 2024-09-15:
To je pořád na stejném místě nicméně je tam informace `ERROR: Failed to find link 08 Profil výroby za den (-A)` což je spojené s portálem - mohu tedy poprosit ještě o poslední obrázky obrazovek z /appdaemon/apps/pnd poslední bude asi několik obrázků **interval-body-08**

waclaw66, 2024-09-15:
Posledni bylo interval-body-07b...
[image]

Zkusil jsem jeste zmensit interval do 1.1.2025 a tentokrat vse probehlo ok. CEZ asi nepocita s rokem 2030 :D

closed 2024-09-15 by waclaw66

ondrejvysek, 2024-09-23:
Diky, dobre vedet, s takto dlouhym intervalem jsem nikdy netestoval a asi dokazu pochopit ze to skonci chybou / trva dlouho. Pouziti puvodniho `wait_for_download` neni na skodu, akorat se starsim chrome driverem se chova dost nevyzpitatelne