ondrejvysek/HomeAssistant-CEZDistribuce-PND #51: AppDaemon TypeError: 'str' object does not support item assignment

Kroupovi, 2024-09-26:
Ahoj celý den se snažím rozchodit integraci a pořád a pořád bojuji. AppDaemon mi píše tuto chybu TypeError: 'str' object does not support item assignment už jsem vše přenstaloval 2x a stále nic nepomáhá pomohl by někdo pls obrázky zašlu obratem děkuji

ondrejvysek, 2024-09-26:
Prosím o komplet výpis chyby z AppDaemon, aby bylo k dohledání ve které části skriptu se chyba nachází + výpis komponent, které jsou vypsány pos restaru AppDaemon. Konec výpisu vypadá takto nějak:
```
trio-websocket     0.11.1
typing_extensions  4.12.2
tzdata             2024.1
urllib3            1.26.18
uvloop             0.17.0
websocket-client   1.8.0
wheel              0.42.0
wsproto            1.2.0
yarl               1.9.4

ChromeDriver Version: ChromeDriver 124.0.6367.78 (a087f2dd364ddd58b9c016ef1bf563d2bc138711-refs/branch-heads/6367@{#954})
```

Kroupovi, 2024-09-26:
Psal jsem na mail ale nevím jako to chodí ještě jsem nic nepotřeboval tak se omlouvam jestli je to špatně. Toto je log nyní . Restartoval jsem a takto to zůstalo.

[a0d7b954_appdaemon_2024-09-26T16-19-24.405Z.log](https://github.com/user-attachments/files/17152349/a0d7b954_appdaemon_2024-09-26T16-19-24.405Z.log)

ondrejvysek, 2024-09-26:
Mailem bohuzel nic nedorazilo. Nevím, zdali tomu logu něco nechybí? není tam "AppDaemon Started" a pak by tam někde mělo být viz obr. Ale nevadí, zkusíme jinak :) když teď spustíte atuomatizaci, prosím o větší detail logu s chybou - aby tam bylo vidět kde ta chyba vznikla
[image]

Kroupovi, 2024-09-26:
odeslal jsem další dva maily... test

Kroupovi, 2024-09-26:
tak toto je log kde to začne cyklovat a log ve složce pnd se neobjeví
[a0d7b954_appdaemon_2024-09-26T16-38-40.660Z.log](https://github.com/user-attachments/files/17152580/a0d7b954_appdaemon_2024-09-26T16-38-40.660Z.log)
děkuji

ondrejvysek, 2024-09-26:
ok, díky
1. nevim kam maily posílate, nic nedorazilo :) , zkuste ondrej@vysek.net (jak pošlete, zde jej smažu)
2. v pnd složce se žádný log nedělá, vše jde do appdaemon logu, mj. je dostupný i v UI - viz obrázek
3. chyba je evidentně při startu AppDaemon, k pnd skriptu to nedojde, co je příčinou netuším, ale můžeme zkusit:
- pošlete /config/appdaemon.yaml
- zkuste znovu vložit skript jako raw https://raw.githubusercontent.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/refs/heads/main/pnd.py
- pokud nezabere, zkusit novou instalaci AppDaemon (je tam ze včera nová verze), po instalaci nedělat žádnou další konfiguraci a zkontrolovat, že AppDaemon správně nastartoval.

Případně - a více po mailu - můžeme zkusit vzdálenou plochu nebo pokud máte HA dostupné z internetu, mohl bych se na to podívat.

[image]

ondrejvysek, 2024-09-26:
Vase maily dorazily, ale ja vam zaslat nemohu, vraci se "Vaši zprávu nešlo doručit. Služba DNS (Domain Name System) ohlásila, že doména příjemce neexistuje." Zkuste kdyztak jiny mail?

ondrejvysek, 2024-10-02:
Zkopírován skript z raw odkazu výše, zachováno správné UTF8 kodování

closed 2024-10-02 by ondrejvysek