ondrejvysek/HomeAssistant-CEZDistribuce-PND #52: chromium zombie proces

waclaw66, 2024-10-11:
Tento doplnek mi bezi v appdaemonu provozovanem v samostatnem kontejneru, ne jako soucast hassos.
Po kazdem behu ze stava, ze process `chromium` (7 instanci) zustane neukoncen a zobrazuje se jako zombie proces. Po nejake dobe jich jsou stovky.
Zjistil jsem, ze je nutne instanci tridy `webdriver.Chrome` ukoncovat nejenom metodou `quit()`, ale i predtim metodou `close()`.

```python
    # Close the browser
    driver.close()
    driver.quit()
```
Viz.
https://stackoverflow.com/questions/38518998/selenium-leaves-behind-running-processes

waclaw66, 2024-10-11:
Bohuzel tohle uplne nepomuze, uprava je trochu slozitejsi, pripravim PR.

ondrejvysek, 2024-10-11:
ok, prip jsem .close() pridal do 0.9.9.2

closed 2025-02-02 by ondrejvysek