ondrejvysek/HomeAssistant-CEZDistribuce-PND #55: SKript nestahuje data

gbazala, 2024-10-22:
Rád bych požádal o pomoc s odhalením chyby. V HA jsem naprostý nováček, snažil jsem se tedy postupovat podle návodu podle nejlepšího vědomí a svědomí. Nakonec po ručním spuštěním skriptu skript sice proběhne, ale nestáhne žádná data. Díky.

Instalace: Rapsberry PI 5

Home Assistant

    Core 2024.10.3
    Supervisor 2024.10.2
    Operating System 13.2
    Frontend 20241002.3

Log:

Downloading numpy-2.1.2-cp311-cp311-musllinux_1_2_aarch64.whl (14.4 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 14.4/14.4 MB 10.5 MB/s eta 0:00:00
Installing collected packages: tzdata, numpy, pandas
Successfully installed numpy-2.1.2 pandas-2.2.3 tzdata-2024.2
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting numpy==1.26.4
  Downloading https://wheels.home-assistant.io/musllinux-index/numpy-1.26.4-cp311-cp311-musllinux_1_2_aarch64.whl (11.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.3/11.3 MB 10.4 MB/s eta 0:00:00
Installing collected packages: numpy
  Attempting uninstall: numpy
    Found existing installation: numpy 2.1.2
    Uninstalling numpy-2.1.2:
      Successfully uninstalled numpy-2.1.2
Successfully installed numpy-1.26.4
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting bs4
  Downloading https://wheels.home-assistant.io/musllinux-index/bs4-0.0.2-py2.py3-none-any.whl.metadata (411 bytes)
Collecting beautifulsoup4 (from bs4)
  Downloading https://wheels.home-assistant.io/musllinux-index/beautifulsoup4-4.12.3-py3-none-any.whl (147 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 147.9/147.9 kB 2.6 MB/s eta 0:00:00
Collecting soupsieve>1.2 (from beautifulsoup4->bs4)
  Downloading https://wheels.home-assistant.io/musllinux-index/soupsieve-2.6-py3-none-any.whl (36 kB)
Downloading https://wheels.home-assistant.io/musllinux-index/bs4-0.0.2-py2.py3-none-any.whl (1.2 kB)
Installing collected packages: soupsieve, beautifulsoup4, bs4
Successfully installed beautifulsoup4-4.12.3 bs4-0.0.2 soupsieve-2.6
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[20:52:33] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-10-22 20:52:35.801742 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-10-22 20:52:35.801897 INFO AppDaemon: Python version is 3.11.10
2024-10-22 20:52:35.801968 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-10-22 20:52:35.802038 INFO AppDaemon: Added log: AppDaemon
2024-10-22 20:52:35.802127 INFO AppDaemon: Added log: Error
2024-10-22 20:52:35.802189 INFO AppDaemon: Added log: Access
2024-10-22 20:52:35.802251 INFO AppDaemon: Added log: Diag
2024-10-22 20:52:35.802323 INFO AppDaemon: Added log: pnd
2024-10-22 20:52:35.920681 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-10-22 20:52:36.124672 INFO HASS: HASS Plugin Initializing
2024-10-22 20:52:36.124833 INFO HASS: HASS Plugin initialization complete
2024-10-22 20:52:36.125127 INFO AppDaemon: Initializing HTTP
2024-10-22 20:52:36.125594 INFO AppDaemon: Using 'ws' for event stream
2024-10-22 20:52:36.127345 INFO AppDaemon: Starting API
2024-10-22 20:52:36.128371 INFO AppDaemon: Starting Admin Interface
2024-10-22 20:52:36.128520 INFO AppDaemon: Starting Dashboards
2024-10-22 20:52:36.135535 INFO HASS: Connected to Home Assistant 2024.10.3
2024-10-22 20:52:36.141244 INFO AppDaemon: App 'pnd' added
2024-10-22 20:52:36.141858 INFO AppDaemon: Found 1 active apps
2024-10-22 20:52:36.141972 INFO AppDaemon: Found 0 inactive apps
2024-10-22 20:52:36.142060 INFO AppDaemon: Found 0 global libraries
2024-10-22 20:52:36.142164 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-10-22 20:52:36.142590 INFO AppDaemon: Running on port 5050
2024-10-22 20:52:36.160834 INFO HASS: Evaluating startup conditions
2024-10-22 20:52:36.163518 INFO HASS: Startup condition met: hass state=RUNNING
2024-10-22 20:52:36.163682 INFO HASS: All startup conditions met
2024-10-22 20:52:36.175449 INFO AppDaemon: Got initial state from namespace default
2024-10-22 20:52:38.146498 INFO AppDaemon: Scheduler running in realtime
2024-10-22 20:52:38.147935 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2024-10-22 20:52:38.148170 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pnd to module import path
2024-10-22 20:52:38.148991 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pnd.py
2024-10-22 20:52:38.153485 WARNING Error: ------------------------------------------------------------
2024-10-22 20:52:38.153703 WARNING Error: Unexpected error loading module: /homeassistant/appdaemon/apps/pnd.py:
2024-10-22 20:52:38.153849 WARNING Error: ------------------------------------------------------------
2024-10-22 20:52:38.156911 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/homeassistant/appdaemon/apps/pnd.py", line 112
    <title>HomeAssistant-CEZDistribuce-PND/pnd.py at main · ondrejvysek/HomeAssistant-CEZDistribuce-PND · GitHub</title>
                                                          ^
SyntaxError: invalid character '·' (U+00B7)

2024-10-22 20:52:38.157111 WARNING Error: ------------------------------------------------------------
2024-10-22 20:52:38.157253 WARNING AppDaemon: Removing associated apps:
2024-10-22 20:52:38.157388 WARNING AppDaemon: pnd
2024-10-22 20:52:38.157783 INFO AppDaemon: App initialization complete

gbazala, 2024-10-25:
Dobrý den! 

Díky za zprávu, o víkendu na to mrknu...

S pozdravem

Gustav Bazala

> Prosím aktualizujte skript z raw souboru [1] , je to stejná chyba jako #49 [2] řádek 112 by měl být prázdný. příp vložte otisk obrazovky jak skript vypadá v editoru. 
 

Links:
------
[1]
https://raw.githubusercontent.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/refs/heads/main/pnd.py
[2]
https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/issues/49
[3]
https://github.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/issues/55#issuecomment-2435798449
[4]
https://github.com/
--=_0b0ac0b30dd6cd9a24c29e23b956be89
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 8pt; font-family: Verdana,Gene=
va,sans-serif'>
<p>Dobr&yacute; den!</p>
<p><br /></p>
<p>D&iacute;ky za zpr&aacute;vu, o v&iacute;kendu na to mrknu...</p>
<div>
<div class=3D"pre" style=3D"margin: 0; padding: 0; font-family: monospace">=
S pozdravem<br /> <br /> <br /> Gustav Bazala<br /> <br /> ________________=
___________<br /> </div>
</div>
<p><br /></p>
<p>Dne 24.10.2024 19:00, ondrejvysek napsal:</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<p><br /></p>
<p dir=3D"auto">Pros&iacute;m aktualizujte skript z <a href=3D"https://raw=
=2Egithubusercontent.com/ondrejvysek/HomeAssistant-CEZDistribuce-PND/refs/h=
eads/main/pnd.py" rel=3D"nofollow">raw souboru</a> , je to stejn&aacute; ch=
yba jako <a class=3D"issue-link js-issue-link" href=3D"https://github.com/o=
ndrejvysek/HomeAssistant-CEZDistribuce-PND/issues/49">#49</a> =C5=99&aacute=
;dek 112 by m=C4=9Bl b&yacute;t pr&aacute;zdn&yacute;. p=C5=99&iacute;p vlo=
=C5=BEte otisk obrazovky jak skript vypad&aacute; v editoru.</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
ndrejvysek/HomeAssistant-CEZDistribuce-PND/issues/55#issuecomment-243579844=
9">view it on GitHub</a>, or <a href=3D"https://github.com/notifications/un=
subscribe-auth/AVRCJKB2QMDNER54UNLPJ2TZ5ERSPAVCNFSM6AAAAABQNF6FNKVHI2DSMVQW=
IX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDIMZVG44TQNBUHE">unsubscribe</a>.<br />You a=
re receiving this because you authored the thread.<img src=3D"https://githu=
b.com/notifications/beacon/AVRCJKE6O3JIVPCKYC4UI53Z5ERSPA5CNFSM6AAAAABQNF6F=
NKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTURF5K3C.gif" =
width=3D"1" height=3D"1" /><span style=3D"color: transparent; font-size: 0;=
 display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0p=
pan>&lt;ondrejvysek/HomeAssistant-CEZDistribuce-PND/issues/55/2435798449</s=
pan><span>@</span><span>github</span><span>.</span><span>com&gt;</span></sp=
an></p>
</blockquote>
</body></html>

--=_0b0ac0b30dd6cd9a24c29e23b956be89--

ondrejvysek, 2024-11-01:
Jak to prosím dopadlo?

gbazala, 2024-11-01:
K testu jsem se dostal až dnes, omlouvám se proto za zpoždění.

Chyba byla opravdu v souboru pnd.py. Původní soubor jsem kopíroval do HA tak, že jsem v návodu klikl pravou myší na soubor a přes Uložit odkaz jako... soubor uložil. Uložený soubor jsem potom přes Upload File v File editoru HA nahrál na požadované místo. Nenapadlo by mne proto, že se bych mohl mít nějakou špatně formátovanou verzi souboru...

Při opravě jsem použil odkaz na neformátovaný text z návodu k opravě a přenesl pomocí CTRL+C a CTRL+V. Na první dobrou potom vše fungovalo. Možná bych toto zmínil v návodu pro podobné LAMY, jako já :-)

Ještě jednou díky, jdu si hrát s grafy.

ondrejvysek, 2024-11-01:
Super, díky za zpětnou vazbu, v dokumentaci jsem upravil odkaz na ten RAW soubor, a info o copy/paste, snad pomůže.

closed 2024-11-01 by ondrejvysek