ondrejvysek/HomeAssistant-CEZDistribuce-PND #56: 10 denní příklad je asi špatně

evlo, 2024-10-27:
Moc ty apex chart nechápu, ale zkusil jsem upravit ten 10 denní příklad na 14 dní a hlavní problém vidím, že to ukazuej za 2024-10-27 spotřebu 24.2kWh, ale ta byla 2024-10-26, za 27 jeste v PND 27. neznáme.

Vedlejší problém, že pro 2024-10-14 chybí datum v té dolní ose

```      - type: grid
        cards:
          - type: custom:apexcharts-card
            stacked: true
            graph_span: 14d
            span:
              end: day
            header:
              show: true
              title: PND Posledních 14 dní
            series:
              - entity: sensor.pnd_data
                name: Spotřeba
                attribute: consumption
                data_generator: |
                  return entity.attributes.pnddate.map((pnd, index) => {
                    return [new Date(pnd).getTime(), entity.attributes.consumption[index]*-1];
                  });
                color: var(--error-color)
                opacity: 0.9
                invert: true
                type: column
                show:
                  datalabels: true
            apex_config:
              dataLabels:
                position: bottom
                dropShadow:
                  enabled: true
              plotOptions:
                bar:
                  dataLabels:
                    position: 'bottom'
            layout_options:
              grid_columns: 8
              grid_rows: 4
            experimental:
              disable_config_validation: true```

![image](https://github.com/user-attachments/assets/99bb2c74-2c48-4727-b6a3-4e0d89dc3d1f)

a pak teda nevím jak otočit ty mínusy v datalabels

ondrejvysek, 2024-10-27:
Jak je uvedeno v popisu:
Využívá senzory sensor.pnd_consumption a sensor.pnd_production které obsahují denní spotřebu resp výrobu za **předchozí den**.

tj dnes (27.10) jsou tam včerejší data (26.10) (dle nastavení automatizace se po půlnoci stáhnou data za předchozí den). Bez zásahu do databáze není možné datum upravit na skutečné.

pokud je potřeba 10 dní protáhnout na 14dní, pak toto `graph_span: 10d` změnit na `graph_span: 14d` z příkladu.

v grafu nevím co má být účelem, nicméně je tam pouze jeden data set, tj. nebude ukazovat spotřebu a výrobu. Druhá věc, co je tam zvláštní je že se násobí *-1 (tím se z toho stane záporné číslo, aby se dostalo dolů), ale o kousek níž je invert: true a to z toho vyrobí opět kladné hodnoty. Pokud je cílem dostat do záporných čísle, pak jedno z výše uvedeného vyhodit.

Ono všechny ty grafy jsou počítané stejně, jenom se liší span (kolik dní v grafu).

closed 2024-10-27 by ondrejvysek

evlo, 2024-10-28:
Aha, když by to někomu dalšímu přišlo divné, nebo já to v budoucnu hledal, tak jsem řádek 799 změnil takto `date_str = [(dt - datetime.timedelta(days = 1)).isoformat() for dt in data_consumption.iloc[:, 0]]`

samozřejmě, že pokud mě někdo bude následovat, tak mu pak nepůjde updatovat bez toho aby zas musel hledat kde to změnit, ale nevím jak to vyřešit v grafu ani v datech (jednoduše), tohle mě přišlo nejjednodušší.

evlo, 2024-10-28:
Hmm, to asi nemá ani středně složitý řešení, protože jsem si říkal, že použiju statistics graph a pnd_consumption, napřed jsem teda udělal recorder purge, ale problém je, že je jedno jaký datum je zapsaný v atributu, protože to zapisuje hodnotu ve chvíli kdy to běží, což je asi vlastnost HA senzoru, že nejde určit datum.