ondrejvysek/HomeAssistant-CEZDistribuce-PND #59: GRAFY problém

K4BlOs, 2025-01-06:
Po delší době jsem zjistil, že mi přestaly fungovat některé grafy (PND 10 dní, Týdenní agregace a Měsíční agregace).

**PND Posledních 10 dní**: graf teď vypadá takto
[image]

kód mám takto (dle default nastavení)
```
type: custom:apexcharts-card
view_layout:
  grid-area: pnd_10_dni
stacked: true
graph_span: 10d
span:
  end: day
header:
  show: true
  title: PND Posledních 10 dní
series:
  - entity: sensor.pnd_data
    name: Výroba
    attribute: production
    data_generator: |
      return entity.attributes.pnddate.map((pnd, index) => {
        return [new Date(pnd).getTime(), entity.attributes.production[index]];
      });
    color: var(--success-color)
    opacity: 0.8
    invert: false
    type: column
  - entity: sensor.pnd_data
    name: Spotřeba
    attribute: consumption
    data_generator: |
      return entity.attributes.pnddate.map((pnd, index) => {
        return [new Date(pnd).getTime(), entity.attributes.consumption[index]];
      });
    color: var(--error-color)
    opacity: 0.8
    invert: true
    type: column
```

přitom sensor.pnd_data obsahuje hodnoty jak v Consumption tak v Production atributech
[image]

**PND Historická Data (Týdenní agregace)**: graf teď vypadá takto. Nejdříve fungoval a pak jakoby se načítala jen jedna stejná hodnota
[image]

kód mám takto (dle default nastavení)
```
type: custom:apexcharts-card
view_layout:
  grid-area: pnd_tyden
stacked: true
graph_span: 1y
span:
  end: week
header:
  show: true
  title: PND Historická Data (Týdenní agregace)
series:
  - entity: sensor.pnd_data
    name: Výroba
    attribute: production
    data_generator: |
      return entity.attributes.pnddate.map((pnd, index) => {
        return [new Date(pnd).getTime(), entity.attributes.production[index]];
      });
    color: var(--success-color)
    opacity: 0.8
    invert: false
    type: column
    group_by:
      func: sum
      duration: 7d
  - entity: sensor.pnd_data
    name: Spotřeba
    attribute: consumption
    data_generator: |
      return entity.attributes.pnddate.map((pnd, index) => {
        return [new Date(pnd).getTime(), entity.attributes.consumption[index]];
      });
    color: var(--error-color)
    opacity: 0.8
    invert: true
    type: column
    group_by:
      func: sum
      duration: 7d
```

**PND Historická Data (Měsíční agregace)**: graf teď vypadá takto. Nejdříve fungoval a pak jakoby se načítala jen jedna stejná hodnota
[image]

kód mám takto (dle default nastavení)
```
type: custom:apexcharts-card
apex_config:
  chart:
    height: 420px
view_layout:
  grid-area: pnd_mesic
stacked: true
graph_span: 1y
span:
  end: day
header:
  show: true
  title: PND Historická Data (Měsíční agregace)
series:
  - entity: sensor.pnd_data
    name: Výroba
    attribute: production
    data_generator: |
      return entity.attributes.pnddate.map((pnd, index) => {
        return [new Date(pnd).getTime(), entity.attributes.production[index]];
      });
    color: var(--success-color)
    opacity: 0.8
    invert: false
    type: column
    group_by:
      func: sum
      duration: 1month
  - entity: sensor.pnd_data
    name: Spotřeba
    attribute: consumption
    data_generator: |
      return entity.attributes.pnddate.map((pnd, index) => {
        return [new Date(pnd).getTime(), entity.attributes.consumption[index]];
      });
    color: var(--error-color)
    opacity: 0.8
    invert: true
    type: column
    group_by:
      func: sum
      duration: 1month
```

Přijde mi, že ten ```data_generator``` teď nějak špatně generuje data z ```sensor.pnd_data```
Nějaký nápad jak grafy opravit?

K4BlOs, 2025-01-06:
Tak jsem pěknej blb. Hned jak jsem dopsal příspěvek tak mě to trklo. Neupravil jsem datum ```DataInterval``` pro následující období. Nyní již vše ok a omlouvám se :-)

closed 2025-01-06 by K4BlOs

ondrejvysek, 2025-01-06:
Mám rád problémy, které se vyřeší samy :) netřeba se omlouvat. Inspirace, zkusím tam přidat senzor na interval pro stahování dat, aby to z toho bylo zjevné a možná mě napadlo i minimum a maximum datum, které to stáhlo

ondrejvysek, 2025-01-06:
Kdyz on ne kazdy tam ma rocni interval. Ale pokud by byl zajem, tak mohu nejak vymyslet logiku neco jako pocatecni datum a pak by to rolovalo po roce

K4BlOs, 2025-01-07:
Za mě by to bylo super 👌