ondrejvysek/HomeAssistant-CEZDistribuce-PND #60: sensor.pnd_data

matomatusov, 2025-01-19:
Ahoj, nevieš ako to ošetriť aby neprekračovala povolenú velikosť 16 384 bajtu? Može to sposobiť problém s výkonom databáze.
Tak ma napadlo, buď:

-  snížiť nejak velikosť atributu
- vylučením atributu z databaze

Logger: homeassistant.components.recorder.db_schema
Zdroj: components/recorder/db_schema.py:658
integrace: Recorder (dokumentace, problémy)
První výskyt: 17. ledna 2025 v 23:05:43 (4 výskyty)
Naposledy logováno: 16:33:09

State attributes for sensor.pnd_data exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored

matomatusov, 2025-01-19:
Ahoj, interval mám nastavený od 1.1 2024 do 31.12 2025. Neviem nájsť cetu k tomu súboru alebo jeho velikosť. Nevieš ako sa dá v HA k tomu dostať?

ondrejvysek, 2025-01-19:
Vše se ukládá do /homeassistant/appdaemon/apps/pnd. Buď můžeš nasdílet debug.zip, který se udělá po každém úspěšném běhu nebo pak jde primárně o soubory range-consumption.csv a range-production.csv.

[image]

ještě možná - jak máš nastavený recorder. pokud není změněno jeho nastavení, tak to moc nevadí, i kdy to bylo víc dat, stahují se pouze 1x za den (chápu problém u senzorů, které to dělají pravidelně) a standardně jsou data pryč do 10ti dní. tj i kdyby to mělo 100K x 10 dní je 1MB max a to fakt problém neudělá. 

Pro klid duše, ano, teoreticky by šlo uložit do externí databáze,... ale jak se k tomu v nativních grafech dostat je problém. Nebo, hodit HA na externí databázi z té souborové, ale. Vše má své výhody a nevýhody.

btw provozuji HA na ProxMox, DB je cca 1.3GB (4000+ devices, 6000+entit) a problém není

matomatusov, 2025-01-19:
Ja ho provozuje na VM v Synology. Na HA mám cca vyhradené 250GB. Velikosť toho súboru mám 
➜  ~ du -k /homeassistant/appdaemon/apps/pnd 
6544    /homeassistant/appdaemon/apps/pnd 

Zistil som cez terminál v kB

ondrejvysek, 2025-01-19:
Kdyžtak velikost jenom těch 2 souborů, tohle bere i obrázky, co se tam ukládají z průběhu. Lze klidně přes file editor. 6.5MB je docela dost, ale to nebudou pouze ty stažené exporty z PND.

matomatusov, 2025-01-19:
[image]

ondrejvysek, 2025-01-19:
To asi sedí, data jsou v grafech zobrazena? Jak to píše že neuloží atribut: State attributes for sensor.pnd_data exceed maximum size of 16384 bytes. This can cause database performance issues; **Attributes will not be stored**

Jak máš nastavený recorder (resp je ve výchozím nastavení). Pokud je změněný, pak by mohlo zabrat 

```
recorder:
  exclude:
    entities:
      - sensor.pnd_data
```

ale nevím jak se pak projeví v datech, zkus.

nicméně jak jsem psal, pokud je recoder ve výchozím stavu, grafy se zobrazují, nestresoval bych se 10dní * 28.8KiB = 280KiB (resp x2 - spotřeba a výroba) víc to nevezme

matomatusov, 2025-01-19:
Ano dáta su v grafoch zobrazena a recorder je v default stave. Ten sensor.pnd_data nemám vylučený z recorderu.

Dik moc za info.

ondrejvysek, 2025-01-19:
ok, jak jsem psal, s touhle hláškou se nedá nic udělat. 
koukni jak dlouho se ti tam ta data ukládají - mělo by to být těch 10dní. Pokud to tak je, opravdu bych se netrápil a s tou hláškou se neučil žít.
Příp zkusit vyloučit senzor z recorderu, ale tam hrozí, že se data neuloží vůbec, je potřeba zkusit.

další možnost - dlouhodobě sledovat vývoj velikosti DB a pak to případně řešit.

pokud bys cokoliv z toho zkusil, dej vědět, zdali zabralo, pomohlo by ostatním.

ondrejvysek, 2025-02-02:
ve verzi 0.9.9.3 z 2.2.2025 sníženo množství ukládaných dat

closed 2025-02-02 by ondrejvysek