ondrejvysek/HomeAssistant-CEZDistribuce-PND #67: Fail při hledání h1_text = "Naměřená data"

martinargalas, 2025-03-14:
Dostávám timeout na řádku 229 v pnd.py. Zprvu jsem myslel, že je to login issue, ale podle screenshotu, který jsem udělal těsně před failem, jsem přihlášený. Vypadá to, že se zobrazuje pouze header a footer a tím pádem není dostupný string "Naměřená data". [error.txt](https://github.com/user-attachments/files/19253139/error.txt)

Mám to rozjeté v Dockeru a tak si říkám, jestli je třeba v containeru ještě něco doinstalovat pro prohlížení webů. Jediné co mě ještě napadlo, že by ta chyba mohla mít něco společného s aktuální hláškou na PND (nefunkční správa upozornění), ale to by ten skript asi nefungoval nikomu (usuzuju z toho že poslední, co je na screenshotu vidět je jejich variable {{ activeImportantMessage }}. V AppDaemon mám jinak přidány všechny system a python packages, jak máš v návodu.

Screen z PND ohledně nefunkční správy upozornění:
[image]

Screen přidaný těsně přes execution řádku 228 v pnd.py:
[image]

Dokázal bys říct, co může být špatně?

ondrejvysek, 2025-03-16:
Ahoj, tuto obrazovku jsem zatím neviděl, není to jenom krátkodobý problém? Pokud by něco chybělo nainstalované, tak by to stránku neotevřelo vůbec (resp nezpracovalo stáhnutá data). řádek 229 kontrolu přítomnost úspěšné otevření portálu, což určitě souvisí s tou hláškou - screenshoty to kdyžtak ukládá samo vždy po vyvolání nějaké interakce do složky /pnd

martinargalas, 2025-03-16:
Ahoj, jestli je to krátkodobý problém nevím. Zkouším to rozjet poprvé. Pokud bys mi potvrdil, že ti to teď taky nenačítá, tak je jasné, že to bude problém na straně PND.

Přijde mi jako kdyby se mi vůbec nenačetl web content. Když jsem se díval na přímo na PND v browser debuggeru, tak tam mají ještě před načtením contentu nějaký javascript, který řeší správu PND messages, ale těžko říct, jestli to má nějaký vliv.

Jinak o těch screenshotech vím, uloží mi to pouze screen 00 a dál se to nedostane. Tak jsem vzal řádek, který ti v pnd.py vytváří screenshoty a vložil ho těsně před ten řádek 229 a dostal jsem to, co jsem přiložil.

ondrejvysek, 2025-03-16:
Na 3 instancích to dnes jede v poho, nemám nyní možnost ověřit běh v dockeru. To spíš, zdali se dnes nerozjelo. Zkus kouknout do diskuzí tu, bylo tu pár lidí, co rozjížděli v dockeru.

martinargalas, 2025-03-16:
Díky. To s odkliknutím zprávy mě taky napadlo, nepomohlo. Zkoušel jsem i zvedat timeout, ale zatím taky nic. V zavřených bugs jsem nic relevantního nenašel.  Ještě nějaký čas tomu dám a uvidím, jestli s tím pohnu.

Přijde mi, že mi něco chybí doinstalovat v containeru, protože je divné, že login page screen  z PND je i s grafikou a pak to, co jsem ti poslal je vyloženě v textovém režimu.

ondrejvysek, 2025-03-16:
To je minimálně zvláštní. možná, kouknout na verzi Chromium, vím, že v minulosti bylo nutné AppDaemon dát na beta kanál, aby se stáhla novější verze Chromia (se starou to nejelo)

martinargalas, 2025-03-16:
Jojo, tak jsem to nakonec rozjel tím, že jsem AppDaemon vzal z dev a nikoli latest. Opravdu pomohla novější verze Chromium. Pak už bylo jen potřeba upravit pár věcí ve skriptu, podle toho jak mám nadefinované cesty k souborům**

docker-compose:
```
services:
  appdaemon:
    container_name: appdaemon
    restart: unless-stopped
    image: acockburn/appdaemon:dev
    privileged: true 
    environment:
      - HA_URL=http://xxx.xxx.xxx.xxx:8123/
      - HA_KEY=xxxxx
    ports:
      - 5050:5050
    volumes:
      - /opt/appdaemon/config:/conf
```

apps.yaml
```
pnd:
  module: pnd
  class: pnd
  log: pnd
  log_level: INFO 
  PNDUserName: peter_argalas
  PNDUserPassword: Gargamel191964
  DataInterval: "21.6.2024 00:00 - 28.2.2025 00:00"
  ELM: "3000013220"
  DownloadFolder: "/conf/pnd"
```

appdaemon.yaml
```
appdaemon:
  latitude: 52.379189
  longitude: 4.899431
  elevation: 2
  time_zone: Europe/Amsterdam
  thread_duration_warning_threshold: 60
  plugins: 
    HASS:
      type: hass
      ha_url: "http://xxx.xxx.xxx.xxx:8123/"
      token: "xxxxxxxx"
      cert_verify: True
  app_dir: /conf/apps
http:
  url: http://0.0.0.0:5050
admin:
api:
hadashboard:
logs:
  pnd:
    name: pnd
    filename: /conf/pnd.log
```

** Jelikož mám v docker-compose definovaný config volume jako "conf", bylo třeba v pnd.py upravit řádky 847 a 848 z:
```
zip_folder("/homeassistant/appdaemon/apps/pnd", "/homeassistant/appdaemon/apps/debug.zip")
shutil.move("/homeassistant/appdaemon/apps/debug.zip", self.download_folder+"/debug.zip")
```
na:
```
zip_folder("/conf/apps/pnd", "/conf/apps/debug.zip")
shutil.move("/conf/apps/debug.zip", self.download_folder+"/debug.zip")
```

closed 2025-03-16 by martinargalas

ondrejvysek, 2025-03-16:
Parada 👍