ondrejvysek/HomeAssistant-CEZDistribuce-PND #68: chyba v dockeru

HellixFix, 2025-03-17:
Omlouvám se, ale nejsem schopen se v issues nějak zorientovat - pokud to tu už třeba je vyřešeno.
Těžce jsem rozjel přes portainer appdaemon v dockeru a po spoustě pokusů se mi objevily entity.
v pnd.py jsem opravil cesty

pnd_script_status je ale pořád running, objevil se soubor 00.png, který obsahuje vyplněnou přihlašovací stránku do PND - jiné stažení soubory ne

už nevím, kde pokračovat

v logu HA není nic
toto je log kontejneru

aiohttp-jinja2     1.5.1
aiosignal          1.3.1
appdaemon          4.4.2
astral             3.2
async-timeout      4.0.2
attrs              25.3.0
bcrypt             4.0.1
beautifulsoup4     4.13.3
bidict             0.22.1
bs4                0.0.2
certifi            2022.12.7
charset-normalizer 3.1.0
deepdiff           6.3.0
exceptiongroup     1.2.2
feedparser         6.0.10
frozenlist         1.3.3
h11                0.14.0
idna               3.4
iso8601            1.1.0
Jinja2             3.1.2
MarkupSafe         2.1.2
multidict          6.0.4
numpy              1.26.4
ordered-set        4.1.0
outcome            1.3.0.post0
packaging          24.2
paho-mqtt          1.6.1
pandas             2.2.3
pid                3.0.4
pip                25.0.1
PySocks            1.7.1
python-dateutil    2.8.2
python-dotenv      1.0.1
python-engineio    4.4.0
python-socketio    5.8.0
pytz               2023.3
PyYAML             6.0
requests           2.28.2
selenium           4.10.0
setuptools         65.5.1
sgmllib3k          1.0.0
six                1.16.0
sniffio            1.3.1
sockjs             0.11.0
sortedcontainers   2.4.0
soupsieve          2.6
tomli              2.0.1
tomli_w            1.0.0
trio               0.29.0
trio-websocket     0.12.2
typing_extensions  4.12.2
tzdata             2025.1
urllib3            1.26.15
uvloop             0.17.0
webdriver-manager  4.0.2
websocket-client   1.5.1
wheel              0.40.0
wsproto            1.2.0
yarl               1.8.2
ChromeDriver Version: ChromeDriver 112.0.5615.165 (c262f36e6b1d711ee42d4fbe1343b49960593f18-refs/branch-heads/5615@{#1297})
2025-03-17 18:51:24.927823 INFO AppDaemon: App initialization complete
2025-03-17 17:51:29: ********************* Starting 0.9.9.3 *********************
2025-03-17 17:51:29: ----------------------------------------------
2025-03-17 17:51:29: Hello from AppDaemon for Portal Namerenych Dat
2025-03-17 17:51:31: Driver Loaded
2025-03-17 17:51:34: Website Opened
2025-03-17 18:52:10.989888 WARNING pnd: ------------------------------------------------------------
2025-03-17 18:52:10.990864 WARNING pnd: Unexpected error in worker for App pnd:
2025-03-17 18:52:10.993479 WARNING pnd: Worker Ags: {'id': 'cbbd874a23244afd8cec11ef5a1174a0', 'name': 'pnd', 'objectid': '2bb3888d8f424a65bd11f51ed3826172', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f3ae1303eb0>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2025-03-17T17:51:29.644733+00:00', 'context': {'id': '01JPJKPTVC4HNZ5EYDV27VH1GP', 'parent_id': '01JPJKPTVBRGJ6NRQVEDGNMN4J', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2025-03-17 18:52:10.993853 WARNING pnd: ------------------------------------------------------------
2025-03-17 18:52:11.024889 WARNING pnd: Traceback (most recent call last):
  File "/usr/src/app/conf/apps/pnd.py", line 228, in run_pnd
    h1_element = wait.until(EC.presence_of_element_located((By.XPATH, f"//h1[contains(text(), '{h1_text}')]")))
  File "/usr/local/lib/python3.10/site-packages/selenium/webdriver/support/wait.py", line 95, in until
    raise TimeoutException(message, screen, stacktrace)
selenium.common.exceptions.TimeoutException: Message: 
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1095, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(app, args["kwargs"]))
  File "/usr/src/app/conf/apps/pnd.py", line 230, in run_pnd
    alert_widget_content = driver.find_element(By.CLASS_NAME, "alertWidget__content").text
  File "/usr/local/lib/python3.10/site-packages/selenium/webdriver/remote/webdriver.py", line 740, in find_element
    return self.execute(Command.FIND_ELEMENT, {"using": by, "value": value})["value"]
  File "/usr/local/lib/python3.10/site-packages/selenium/webdriver/remote/webdriver.py", line 346, in execute
    self.error_handler.check_response(response)
  File "/usr/local/lib/python3.10/site-packages/selenium/webdriver/remote/errorhandler.py", line 245, in check_response
    raise exception_class(message, screen, stacktrace)
selenium.common.exceptions.NoSuchElementException: Message: no such element: Unable to locate element: {"method":"css selector","selector":".alertWidget__content"}
  (Session info: headless chrome=112.0.5615.165); For documentation on this error, please visit: https://www.selenium.dev/documentation/webdriver/troubleshooting/errors#no-such-element-exception
2025-03-17 18:52:11.025387 WARNING pnd: ------------------------------------------------------------
2025-03-17 18:52:11.169197 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 42.168524 seconds (limit=10)
2025-03-17 18:52:21.063314 INFO AppDaemon: Client disconnection from Admin Client
2025-03-17 18:52:21.434773 INFO AppDaemon: New client Admin Client connected

ondrejvysek, 2025-03-17:
Koukněte prosím na #67 přijde mi že jsou to stejné podmínky (docker) a stejné místo chyby (+/-228) - řešení tam je použití /dev appdaemonu, kde je novější chrome driver / chromium. Bohužel nemám možnost (zatím) popsat přesný návod / řešení problémů mimo HA

matomatusov, 2025-03-17:
Ahoj @HellixFix , podarilo sa ti to rozchodiť? mám samý problém po updatu HA.

ondrejvysek, 2025-03-17:
@matomatusov tu to ovsem nepise zadnou chybu?

HellixFix, 2025-03-17:
Teď jsem rozjel dev verzi appdaemon, ovšem musel jsem tam ručně doinstalovat skoro všechno. Dokonce už se začaly stahovat soubory (úspěch!!), ale zaseklo se to a nedokončilo.

Package            Version
------------------ -----------
aiohappyeyeballs   2.4.4
aiohttp            3.11.12
aiohttp-jinja2     1.6
aiosignal          1.3.2
appdaemon          4.5.0
astral             3.2
async-timeout      5.0.1
attrs              25.1.0
bcrypt             4.2.1
beautifulsoup4     4.13.3
bidict             0.23.1
bs4                0.0.2
certifi            2025.1.31
charset-normalizer 3.4.1
deepdiff           8.2.0
feedparser         6.0.11
frozenlist         1.5.0
h11                0.14.0
idna               3.10
iso8601            2.1.0
Jinja2             3.1.5
MarkupSafe         3.0.2
multidict          6.1.0
numpy              2.2.4
orderly-set        5.3.0
outcome            1.3.0.post0
paho-mqtt          2.1.0
pandas             2.2.3
pid                3.0.4
pip                25.0.1
propcache          0.2.1
PySocks            1.7.1
python-dateutil    2.9.0.post0
python-engineio    4.11.2
python-socketio    5.11.4
pytz               2025.1
PyYAML             6.0.2
requests           2.32.3
selenium           4.29.0
sgmllib3k          1.0.0
simple-websocket   1.1.0
six                1.17.0
sniffio            1.3.1
sockjs             0.13.0
sortedcontainers   2.4.0
soupsieve          2.6
tomli              2.2.1
tomli_w            1.2.0
trio               0.29.0
trio-websocket     0.12.2
typing_extensions  4.12.2
tzdata             2025.1
urllib3            2.3.0
uvloop             0.21.0
websocket-client   1.8.0
wsproto            1.2.0
yarl               1.18.3
ChromeDriver Version: ChromeDriver 134.0.6998.88 (7e3d5c978c6d3a6eda25692cfac7f893a2b20dd0-refs/branch-heads/6998@{#1898})
2025-03-17 21:54:32.687771 INFO AppDaemon: App initialization complete
2025-03-17 21:54:36.876799 INFO AppDaemon: New client Admin Client connected
2025-03-17 20:55:03: ********************* Starting 0.9.9.3 *********************
2025-03-17 21:55:03.919936 INFO AppDaemon: pnd: Entity binary_sensor.pnd_running created in namespace: default
2025-03-17 21:55:03.933350 INFO AppDaemon: pnd: Entity sensor.pnd_script_status created in namespace: default
2025-03-17 20:55:03: ----------------------------------------------
2025-03-17 20:55:03: Hello from AppDaemon for Portal Namerenych Dat
2025-03-17 20:55:13: Driver Loaded
2025-03-17 20:55:21: Website Opened
2025-03-17 20:55:45: H1 tag with text 'Naměřená data' is present.
2025-03-17 20:55:45: Modal dialog not found. Continuing without closing modal.
2025-03-17 21:55:47.889525 INFO AppDaemon: pnd: Entity sensor.pnd_app_version created in namespace: default
2025-03-17 20:55:47: App Version: 2.4.2
2025-03-17 20:55:55: Rychla Sestava selected successfully!
2025-03-17 20:55:57: Selecting ELM '3000005061'
2025-03-17 20:55:57: Valid ELM numbers 'ELM 3000005061'
2025-03-17 20:56:01: ELM Status: ELM 3000005061 - 3000005061
2025-03-17 20:56:01: Iteration 0: Vyhledat Button NOT disabled
2025-03-17 20:56:01: Device ELM '3000005061' selected successfully!
2025-03-17 20:56:04: Button 'Vyhledat data' clicked successfully!
2025-03-17 20:56:09: 07 Profil spotřeby za den (+A)
2025-03-17 20:56:18: Downloading CSV file for 07 Profil spotřeby za den (+A)
2025-03-17 21:56:24.207018 WARNING pnd: ------------------------------------------------------------
2025-03-17 21:56:24.207470 WARNING pnd: Unexpected error in worker for App pnd:
2025-03-17 21:56:24.207788 WARNING pnd: Worker Ags: {'id': 'b4b75b778efe49639368def80b33fe84', 'name': 'pnd', 'objectid': 'f0753e6ab6e441af80a9e99a29594d48', 'type': 'event', 'event': 'run_pnd', 'function': <bound method pnd.run_pnd of <pnd.pnd object at 0x7f6083c26660>>, 'data': {'metadata': {'origin': 'LOCAL', 'time_fired': '2025-03-17T20:55:03.900981+00:00', 'context': {'id': '01JPJY6YYWGSZXSTNX04TJK3P8', 'parent_id': '01JPJY6YYVRGB0FHMBRKNQBW4S', 'user_id': None}}}, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
2025-03-17 21:56:24.208075 WARNING pnd: ------------------------------------------------------------
2025-03-17 21:56:24.240493 WARNING pnd: Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/appdaemon/threading.py", line 1169, in worker
    funcref(args["event"], data, self.AD.events.sanitize_event_kwargs(
    ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        app, args["kwargs"]))
        ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/app/conf/apps/pnd.py", line 490, in run_pnd
    os.rename(downloaded_file, new_filename)
    ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'conf/apps/pnd/pnd_export.csv' -> 'conf/apps/pnd/daily-consumption.csv'
2025-03-17 21:56:24.240957 WARNING pnd: ------------------------------------------------------------
2025-03-17 21:56:24.242854 WARNING AppDaemon: Excessive time spent in callback 'run_pnd() in pnd', Thread 'thread.thread-0' - now complete after 81.242232 seconds (limit=10)

HellixFix, 2025-03-18:
ten CSV se stahoval do /root/Downloads, tak jsem v kódu napevno přepsal cestu a už to jede

nevím, co jsem dělal špatně při přechodu z last na dev, ale nejelo nic - ani kompilace atd.... tak jsem chyby asi na5x prohnal Claude a postupně se vše aktualizovalo/stáhlo a pak se teprve rozjelo

closed 2025-06-23 by ondrejvysek