ondrejvysek/HomeAssistant-CEZDistribuce-PND #69: AppDaemon v4.4.2 v dockeru, chyba

GRICA23, 2025-04-14:
Ahoj. Mám HA a AppDaemon v dockeru a skrupt hlásí chyby, viz příloha.
Děkuji za pomoc.

[pnd.zip](https://github.com/user-attachments/files/19730368/pnd.zip)

foxlg, 2025-04-17:
Mam to ted stejny.. pri prvnim pusteni vse proslo a pri opakovanem hlasi stejnou hlasku

ondrejvysek, 2025-04-17:
Nejsem nyni u PC abych otestoval/zjistil vic. prosim kouknete na nekterey predchozi uzavreny, vypada hako stejny problem. Pokud ano - zkuste pouzit dev vetev, kde je novejsi chrome driver

foxlg, 2025-04-17:
Tak jsem prosel closed issues a je to neco jineho.. imho je neco zde: `2025-04-17 08:38:09: No cookie banner found` .. ale zkusim to asi zitra pustit jen ze stroje rucne.. aby se dalo debugovat.. jeste jsem nemel tu cest si s tim pracovat.. tak uvidime kdo bude rychlejsi :D

foxlg, 2025-04-17:
> Nejsem nyni u PC abych otestoval/zjistil vic. prosim kouknete na nekterey predchozi uzavreny, vypada hako stejny problem. Pokud ano - zkuste pouzit dev vetev, kde je novejsi chrome driver

Tak jsem prosel closed issues a je to neco jineho.. imho je neco zde: 2025-04-17 08:38:09: No cookie banner found .. ale zkusim to asi zitra pustit jen ze stroje rucne.. aby se dalo debugovat.. jeste jsem nemel tu cest si s tim pracovat.. tak uvidime kdo bude rychlejsi :D

foxlg, 2025-04-17:
> Nejsem nyni u PC abych otestoval/zjistil vic. prosim kouknete na nekterey predchozi uzavreny, vypada hako stejny problem. Pokud ano - zkuste pouzit dev vetev, kde je novejsi chrome driver

Zkusil jsem tedy dev vetev a porad je to stejny.. jina v readme v devu nejsou pripadne vsechny zavislosti na obrazku a ani jsem je nevidel pak v textu ;) (bs4 a numpy==1.26.4)

foxlg, 2025-04-24:
Tak se to zotavilo. Nevim cim to bylo, ale ted mi to jede az na par sensoru `sensor.pnd_production2consumptionfloor` a `sensor.pnd_production2consumptionfull`. Mam ale aktualne dev script, tak je mozny ze se v tom ted lisi.. update ale ted probiha v poradku.