ondrejvysek/HomeAssistant-CEZDistribuce-PND #74: Nesprávný formát dátum a čas

matomatusov, 2025-08-07:
Ahoj,
po updatu HA na 2025.8.0 asi problém v tom, že skript (v AppDaemon aplikácii pnd) sa pokúša spracovať dátum a čas v nesprávnom formáte.

2025-08-07 21:43:16.402608 ERROR pnd: =====  pnd.run_pnd for pnd  ===============================================
2025-08-07 21:43:16.403800 ERROR pnd: EventCallbackFail: Scheduled callback failed for app 'pnd'
2025-08-07 21:43:16.405167 ERROR pnd:   args: ('run_pnd', {'metadata': {'time_fired': '2025-08-07T19:41:50.305272+00:00', 'origin': 'LOCAL', 'context': {'id': '01K230VNB0CZB1TR9XX2TG3FPN', 'parent_id': '01K230VNAZ951KMGKQFFT3X433', 'user_id': None}}}, {'__thread_id': 'thread-0'})
2025-08-07 21:43:16.406346 ERROR pnd:   ValueError: time data '06.08.2025 24:23:59' does not match format '%d.%m.%Y %H:%M'
2025-08-07 21:43:16.409691 ERROR pnd:     File "/usr/lib/python3.12/site-packages/appdaemon/threads.py", line 1080, in safe_callback
2025-08-07 21:43:16.410605 ERROR pnd:       funcref()
2025-08-07 21:43:16.411200 ERROR pnd:     File "/homeassistant/appdaemon/apps/pnd.py", line 569, in run_pnd
2025-08-07 21:43:16.411367 ERROR pnd:       date_consumption_obj = datetime.datetime.strptime(date_consumption_str, date_format)
2025-08-07 21:43:16.411535 ERROR pnd:                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-08-07 21:43:16.411695 ERROR pnd:     File "/usr/lib/python3.12/_strptime.py", line 653, in _strptime_datetime
2025-08-07 21:43:16.411873 ERROR pnd:       tt, fraction, gmtoff_fraction = _strptime(data_string, format)
2025-08-07 21:43:16.412021 ERROR pnd:                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-08-07 21:43:16.412164 ERROR pnd:     File "/usr/lib/python3.12/_strptime.py", line 432, in _strptime
2025-08-07 21:43:16.412300 ERROR pnd:       raise ValueError("time data %r does not match format %r" %
2025-08-07 21:43:16.412460 ERROR pnd: ===========================================================================
2025-08-07 21:43:16.414814 WARNING AppDaemon: Excessive time spent in callback pnd.run_pnd for pnd. Thread entity: 'thread.thread-0' - now complete after 01:26 (limit=01:00)
2025-08-07 22:13:50.159874 WARNING AppDaemon: Excessive time spent in utility loop: 2.2s, 95.996ms in check_app_updates(), 2.1s in other

waclaw66, 2025-08-08:
> možná ještě dodám, že se mi tak nějak sám od sebe rozpadl měsíční sloupcový graf... s tím se mi zatím nepodařilo udělat nic mám dojem, že to asi nezvládá počet dnů v měsíci, speciálně v únoru atd.

ahoj, zkus misto `date.setDate(2);` dat `date.setDate(3);`

waclaw66, 2025-08-08:
ohledne toho padani, vypada, ze CEZ zacal do csv vystupu davat i vteriny a `date_consumption_str.replace("00:00", "23:59")` nefunguje nyni sprave, jelikoz vytvori neplatny cas `24:23:56`

foxlg, 2025-08-10:
Diky. Skvele funguje 👍

closed 2025-08-13 by ondrejvysek