ondrejvysek/HomeAssistant-CEZDistribuce-PND #77: Po aktualizaci HA na 2025.8.3 se nevytvoří senzory pro grafy nebo se změnila stránka PND

zika-martin, 2025-08-25:
Ahoj, netuším příčinu, ale myslím si, že po poslední aktualizaci HA se mi u 0.9.9.4 nevytvoří senzory.
Děkuji za radu
Martin

Log:
2025-08-25 22:44:08.036533 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:08.571472 INFO AppDaemon: Stopping apps from namespace 'default' because the plugin failed
2025-08-25 22:44:08.573604 INFO AppDaemon: Stopping apps: {'pnd'}
2025-08-25 22:44:08.574908 INFO AppDaemon: Calling terminate() for 'pnd'
2025-08-25 22:44:08: >>>>>>>>>>>> PND Terminate
2025-08-25 22:44:13.066142 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:18.074665 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:23.217036 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:28.469934 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:33.476491 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:38.480900 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:43.485333 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:48.490092 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:53.496938 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2025-08-25 22:44:58.552755 INFO HASS: Connected to Home Assistant 2025.8.3 with aiohttp websocket
2025-08-25 22:44:58.557422 INFO HASS: Authenticated to Home Assistant 2025.8.3
2025-08-25 22:44:58.561395 INFO HASS: Waiting for Home Assistant to start
2025-08-25 22:45:35.609490 INFO HASS: Home Assistant fully started after 00:37
2025-08-25 22:45:36.319463 INFO HASS: Completed initialization in 00:38
2025-08-25 22:45:36.619021 INFO AppDaemon: Processing restart for plugin namespace 'default'
2025-08-25 22:45:36.621081 INFO AppDaemon: Starting apps: {'pnd'}
2025-08-25 22:45:36.667153 INFO AppDaemon: Calling initialize() for pnd
2025-08-25 22:45:36: >>>>>>>>>>>> PND Initialize
System Information:
Platform: Linux
Platform Release: 6.12.34-haos-raspi
Platform Version: #1 SMP PREEMPT Wed Aug 13 09:00:58 UTC 2025
Architecture: aarch64
Processor: 
Python Version: 3.12.11
Installed Python Modules:
Package            Version
------------------ -----------
aiohappyeyeballs   2.6.1
aiohttp            3.11.18
aiohttp-jinja2     1.6
aiosignal          1.4.0
annotated-types    0.7.0
appdaemon          4.5.11
astral             3.2
async-timeout      5.0.1
attrs              25.3.0
bcrypt             4.2.1
beautifulsoup4     4.13.4
bidict             0.23.1
bs4                0.0.2
certifi            2025.8.3
charset-normalizer 3.4.2
deepdiff           8.2.0
feedparser         6.0.11
frozenlist         1.7.0
h11                0.16.0
idna               3.10
iso8601            2.1.0
Jinja2             3.1.6
MarkupSafe         3.0.2
multidict          6.6.3
numpy              1.26.4
orderly-set        5.5.0
outcome            1.3.0.post0
packaging          25.0
paho-mqtt          2.1.0
pandas             2.3.1
pid                3.0.4
pip                25.1.1
[... 106 lines cut ...]
2025-08-25 23:51:41.168810 ERROR pnd:   args: ('run_pnd', {'metadata': {'time_fired': '2025-08-25T21:51:01.495720+00:00', 'context': {'id': 'XXXXX', 'parent_id': None, 'user_id': 'XXXXX'}, 'origin': 'LOCAL'}}, {'__thread_id': 'thread-0'})
2025-08-25 23:51:41.189518 ERROR pnd:   StaleElementReferenceException: Message: stale element reference: stale element not found
  (Session info: chrome=139.0.7258.66); For documentation on this error, please visit: https://www.selenium.dev/documentation/webdriver/troubleshooting/errors#staleelementreferenceexception
2025-08-25 23:51:41.221746 ERROR pnd:     File "/usr/lib/python3.12/site-packages/appdaemon/threads.py", line 1080, in safe_callback
2025-08-25 23:51:41.222947 ERROR pnd:       funcref()
2025-08-25 23:51:41.223719 ERROR pnd:     File "/homeassistant/appdaemon/apps/pnd.py", line 293, in run_pnd
2025-08-25 23:51:41.224854 ERROR pnd:       body.screenshot(self.download_folder+"/02.png")
2025-08-25 23:51:41.225866 ERROR pnd:     File "/usr/lib/python3.12/site-packages/selenium/webdriver/remote/webelement.py", line 512, in screenshot
2025-08-25 23:51:41.227021 ERROR pnd:       png = self.screenshot_as_png
2025-08-25 23:51:41.228164 ERROR pnd:             ^^^^^^^^^^^^^^^^^^^^^^
2025-08-25 23:51:41.229464 ERROR pnd:     File "/usr/lib/python3.12/site-packages/selenium/webdriver/remote/webelement.py", line 486, in screenshot_as_png
2025-08-25 23:51:41.230613 ERROR pnd:       return b64decode(self.screenshot_as_base64.encode("ascii"))
2025-08-25 23:51:41.232265 ERROR pnd:                        ^^^^^^^^^^^^^^^^^^^^^^^^^
2025-08-25 23:51:41.233644 ERROR pnd:     File "/usr/lib/python3.12/site-packages/selenium/webdriver/remote/webelement.py", line 472, in screenshot_as_base64
2025-08-25 23:51:41.234517 ERROR pnd:       return self._execute(Command.ELEMENT_SCREENSHOT)["value"]
2025-08-25 23:51:41.235614 ERROR pnd:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-08-25 23:51:41.236959 ERROR pnd:     File "/usr/lib/python3.12/site-packages/selenium/webdriver/remote/webelement.py", line 573, in _execute
2025-08-25 23:51:41.238210 ERROR pnd:       return self._parent.execute(command, params)
2025-08-25 23:51:41.239112 ERROR pnd:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-08-25 23:51:41.239931 ERROR pnd:     File "/usr/lib/python3.12/site-packages/selenium/webdriver/remote/webdriver.py", line 458, in execute
2025-08-25 23:51:41.241379 ERROR pnd:       self.error_handler.check_response(response)
2025-08-25 23:51:41.242578 ERROR pnd:     File "/usr/lib/python3.12/site-packages/selenium/webdriver/remote/errorhandler.py", line 232, in check_response
2025-08-25 23:51:41.243456 ERROR pnd:       raise exception_class(message, screen, stacktrace)
2025-08-25 23:51:41.244675 ERROR pnd: ===========================================================================
2025-08-25 23:51:41.252360 WARNING AppDaemon: Excessive time spent in callback pnd.run_pnd for pnd. Thread entity: 'thread.thread-0' - now complete after 00:40 (limit=10.0s)

zika-martin, 2025-09-05:
Ahoj, omlouvám se za prodlevu, trable v práci.
Hodil jsem verzi 0.9.9.5 do HA, všechno zrestartoval, ale zdá se, že je tam nyní jiná chyba. Senzory se nevytvořily a soubor v /homeassistant/appdaemon/apps/pnd/debug.zip není vytvořen.

2025-09-06 00:39:11: ********************* Starting 0.9.9.5 *********************
2025-09-06 00:39:11: ----------------------------------------------
2025-09-06 00:39:11: Hello from AppDaemon for Portal Namerenych Dat
2025-09-06 00:39:13: Driver Loaded
2025-09-06 00:39:16: Website Opened
2025-09-06 00:39:21: ERROR: Failed to enter login details or find and click the login button
2025-09-06 00:39:21.454480 ERROR pnd: =====  pnd.run_pnd for pnd  ===============================================
2025-09-06 00:39:21.455395 ERROR pnd: EventCallbackFail: Scheduled callback failed for app 'pnd'
2025-09-06 00:39:21.456231 ERROR pnd:   args: ('run_pnd', {'metadata': {'origin': 'LOCAL', 'context': {'id': '01K4E0H8C4JZSEFY5JPF2CGC2D', 'parent_id': None, 'user_id': 'd10d8b21c5ff4c65b0a28d064b5cbf15'}, 'time_fired': '2025-09-05T22:39:11.753481+00:00'}}, {'__thread_id': 'thread-0'})
2025-09-06 00:39:21.458011 ERROR pnd:   Exception: Failed to find or click the login button
2025-09-06 00:39:21.466764 ERROR pnd:     File "/usr/lib/python3.12/site-packages/appdaemon/threads.py", line 1080, in safe_callback
2025-09-06 00:39:21.467509 ERROR pnd:       funcref()
2025-09-06 00:39:21.468226 ERROR pnd:     File "/homeassistant/appdaemon/apps/pnd.py", line 239, in run_pnd
2025-09-06 00:39:21.468792 ERROR pnd:       raise Exception("Failed to find or click the login button")
2025-09-06 00:39:21.469366 ERROR pnd: ===========================================================================
2025-09-06 00:39:21.473183 WARNING AppDaemon: Excessive time spent in callback pnd.run_pnd for pnd. Thread entity: 'thread.thread-0' - now complete after 10.5s (limit=10.0s)

ondrejvysek, 2025-09-06:
Teď bych počkal do 8.9. 6:00. Script havaruje na tom, že se nemůže přihlásit.

[image]

TesaTesa, 2025-09-08:
Funguje ! díky moc.

Nielm75, 2025-09-22:
Zdravím, chtěl bych se přidat s dotazem. Po aktualizaci HA se mi už delší dobu nevytváří sensory a i když jsem to procházel dle dokumentace, tak stále stejný výsledek. Je možno poradit?
debug.zip se nevytvoří, tak jako i sensory. Předtím vše fungovalo, ale po aktualizaci HA již nikoliv.
Díky.

2025-09-22 19:11:27: ********************* Starting 0.9.9.3 *********************
2025-09-22 19:11:27: ----------------------------------------------
2025-09-22 19:11:27: Hello from AppDaemon for Portal Namerenych Dat
2025-09-22 19:11:29: Driver Loaded
2025-09-22 19:11:36: Website Opened
2025-09-22 19:11:56: H1 tag with text 'Naměřená data' is present.
2025-09-22 19:11:57: Modal dialog not found. Continuing without closing modal.
2025-09-22 19:11:59: App Version: 2.4.6
2025-09-22 19:12:07: Rychla Sestava selected successfully!
2025-09-22 19:12:09: Selecting ELM '3000013924'
2025-09-22 19:12:09: Valid ELM numbers 'ELM 10110942, ELM 3000013924, ELM 10110942, ELM 3000013924'
2025-09-22 19:12:13: ELM Status:  - 3000013924
2025-09-22 19:12:13: Iteration 0: Vyhledat Button IS disabled
2025-09-22 19:12:18: ELM Status: ELM 3000013924 - 3000013924
2025-09-22 19:12:18: Iteration 1: Vyhledat Button NOT disabled
2025-09-22 19:12:18: Device ELM '3000013924' selected successfully!
2025-09-22 19:12:22: Button 'Vyhledat data' clicked successfully!
2025-09-22 19:12:26: 07 Profil spotřeby za den (+A)
2025-09-22 19:12:38: Downloading CSV file for 07 Profil spotřeby za den (+A)
2025-09-22 19:12:44: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-consumption.csv
2025-09-22 19:12:45: 08 Profil výroby za den (-A)
2025-09-22 19:12:54: Downloading CSV file for 08 Profil výroby za den (-A)
2025-09-22 19:13:00: File downloaded and saved as: /homeassistant/appdaemon/apps/pnd/daily-production.csv
2025-09-22 19:13:00: All Done - DAILY DATA DOWNLOADED
2025-09-22 19:13:00.532876 ERROR pnd: =====  pnd.run_pnd for pnd  ===============================================
2025-09-22 19:13:00.533498 ERROR pnd: EventCallbackFail: Scheduled callback failed for app 'pnd'
2025-09-22 19:13:00.533818 ERROR pnd:   args: ('run_pnd', {'metadata': {'origin': 'LOCAL', 'time_fired': '2025-09-22T17:11:27.931024+00:00', 'context': {'id': '01K5S6HCDTFWYDVFR8Y4C9RVSW', 'parent_id': '01K5S6HCDSCKWCDNBEYYXK64V3', 'user_id': None}}}, {'__thread_id': 'thread-0'})
2025-09-22 19:13:00.540673 ERROR pnd:   ValueError: time data '21.09.2025 24:23:59' does not match format '%d.%m.%Y %H:%M'
2025-09-22 19:13:00.569845 ERROR pnd:     File "/usr/lib/python3.12/site-packages/appdaemon/threads.py", line 1080, in safe_callback
2025-09-22 19:13:00.570371 ERROR pnd:       funcref()
2025-09-22 19:13:00.577888 ERROR pnd:     File "/homeassistant/appdaemon/apps/pnd.py", line 569, in run_pnd
2025-09-22 19:13:00.579945 ERROR pnd:       latest_production_entry = data_production.iloc[-1]  # Get the last row, assuming the data is appended daily
2025-09-22 19:13:00.583873 ERROR pnd:                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-09-22 19:13:00.585455 ERROR pnd:     File "/usr/lib/python3.12/_strptime.py", line 653, in _strptime_datetime
2025-09-22 19:13:00.585712 ERROR pnd:       tt, fraction, gmtoff_fraction = _strptime(data_string, format)
2025-09-22 19:13:00.585937 ERROR pnd:                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2025-09-22 19:13:00.586187 ERROR pnd:     File "/usr/lib/python3.12/_strptime.py", line 432, in _strptime
2025-09-22 19:13:00.586464 ERROR pnd:       raise ValueError("time data %r does not match format %r" %
2025-09-22 19:13:00.586698 ERROR pnd: ===========================================================================
2025-09-22 19:13:00.592023 WARNING AppDaemon: Excessive time spent in callback pnd.run_pnd for pnd. Thread entity: 'thread.thread-0' - now complete after 01:34 (limit=01:00)